alutov/ESP32-R4sGate-for-Redmond #2: Можно ли добавить MQTT Discovery?

panferovaleksey, 2020-07-16:
Home Assistant поддерживает автоматическое обнаружение и конфигурирование MQTT устройств.
Подробнее: https://www.home-assistant.io/docs/mqtt/discovery/
Вроде бы это просто публикация json с метаданными в определенный MQTT топик.

alutov, 2020-07-22:
ХА не пользуюсь, у меня iobroker.  Точить прошивку именно под ХА пока не планирую, тем более там есть своя интеграция.

vring0, 2020-10-07:
> ХА не пользуюсь, у меня iobroker. Точить прошивку именно под ХА пока не планирую, тем более там есть своя интеграция.
> Upd 07.10.2020. Много просят, попробую таки сделать. Но пока знакомлюсь с самим Hass и изучаю discovery в нем. Водяного нагревателя вроде бы нет в mqtt, есть варианты через свитч, либо климат.

Желательно сделать водный нагреватель, чтобы сущность чайника нормально интегрировалась в Алису. Пока я вижу вариант, это дополнительная интеграция которая работает по MQTT и создает сущность чайника. Вот документация https://developers.home-assistant.io/docs/core/entity/water-heater/ , есть готовая реализация тут https://github.com/mavrikkk/ha_kettler/blob/master/custom_components/ready4sky/water_heater.py только заменить функции BT на MQTT.

alutov, 2020-10-07:
Думаю вот что. Сделаю все, что позволяет ассистент по Mqtt из самой esp32. Скрипты и прочее на самом ассистенте писать нет ни времени, ни навыков.

runnerfanspb, 2020-10-24:
обновил прошивку. все залетело в HA само собой. спасибо огромное

danilovvs, 2020-12-15:
> обновил прошивку. все залетело в HA само собой. спасибо огромное

А какие настройки в setting на шлюзе ставил?

closed 2021-09-24 by alutov