alutov/ESP32-R4sGate-for-Redmond #7: RK-M171S switched to boil itself unexpectedly

slydiman, 2020-09-16:
Подключил RK-M171S в Home Assistant, всё шикарно, но... жена заставила убрать в целях безопасности.

При установке чайника обратно на подставку часто (примерно 1 раз из 3) включался подогрев или кипячение - это напрягало.
В Home Assistant нет никаких автоматизаций для чайника, были только графики и 2 MQTT свича - кипячение и подогрев до 85 градусов. Видимо это HA пытался вернуть последнее состояние. Есть идеи как это предотвратить? Снял чайник - всё должно сброситься! 

Но самое главное - чайник скипел 2 часа назад, никто ничего не трогал, сидим на кухне и вдруг включается кипячение, само!
А если бы это было без нашего присутствия когда чайник был пустой или почти пустой?!
Жалко, но пришлось отключить.
С этим можно что-то сделать? Логи какие-нибудь?

alutov, 2020-09-16:
Проблема скорее всего с сервером mqtt. У меня iobroker со встроенным сервером,mqtt, подобного не замечено. С ХА не знаком. Вот обсуждение на мажордоме: https://mjdm.ru/forum/viewtopic.php?f=8&t=5501&sid=de6b1e2b43f25c8d9ae9af5673ee9417&start=140

slydiman, 2020-09-19:
По непонятной причине очевидно Home Assistant после возобновления связи с чайником в некоторых случаях по собственной инициативе пытался "восстановить" состояние переключателя, то есть включал повторно кипячение или нагрев.
Переписал конфиг Home Assistant, чтобы это был не switch со своим состоянием, а отдельно состояние и просто скрипты boil, heat, off, которые прикрутил к кнопкам в UI. Пока проблема больше не проявлялась. 
Извините за беспокойство.

closed 2020-09-19 by slydiman

alutov, 2020-09-19:
Re: [alutov/ESP32-R4sGate-for-Redmond] RK-M171S switched to boil itself unexpectedly (#7)

Здравствуйте, Dmitry.

Вы писали 19 сентября 2020 г., 20:08:24:

По непонятной причине очевидно Home Assistant после возобновления связи с чайником в некоторых случаях по собственной инициативе пытался "восстановить" состояние переключателя, то есть включал повторно кипячение или нагрев.
Переписал конфиг Home Assistant, чтобы это был не switch со своим состоянием, а отдельно состояние и просто скрипты boil, heat, off, которые прикрутил к кнопкам в UI. Пока проблема больше не проявлялась.
Извините за беспокойство.
—

Нормально все, известная уже проблема с ХА.)))
С уважением,
 Лутов                          mailto:alutov@bk.ru