alutov/ESP32-R4sGate-for-Redmond #8: [Feature request] Порт в интеграцию ESPHome

HepoH3, 2020-09-25:
Было бы очень здорово, если бы была возможность использовать прошивку совместно с ESP-Home. В этом случае, прошивка бы получила очень большую гибкость. Поскольку интеграция поддерживает вставки на c++, со стороны кажется что идея очень даже реальна.

runnerfanspb, 2020-09-28:
поддерживаю данную идею, это было бы огонь огненный

alutov, 2020-10-03:
Я использую Iobroker, люблю "запах сероводорода"(с). Mqtt выбрал как наиболее распространенный интерфейс Iot. С Home Assistant не сталкивался, а потому делать прошивку исключительно под него native API пока не планирую. Можно было бы, конечно, сделать под него discovery, но дело в том, что он сам не поддерживает водонагреватели по Mqtt. Ble сам по себе жрет много ресурсов esp32, так что идея вставки кода в ESP-Home вряд ли реальна. Проще запросить поддержку Redmond в самом ESP-Home.

closed 2020-10-03 by alutov