alutov/ESP32-R4sGate-for-Redmond #9: Redmond SkyBaker

dmamontov, 2020-10-09:
Интересно, есть ли шанс подключить redmond skybaker? может есть у кого инструкция на разбор протокола?

alutov, 2020-10-09:
Есть конечно. Только протокола пока не встречал. И что это за устройство такое? Но можно снимать самому и потом разбираться. Сам это пока не делал, подсказать не могу.

dmamontov, 2020-10-09:
> Есть конечно. Только протокола пока не встречал. И что это за устройство такое? Но можно снимать самому и потом разбираться. Сам это пока не делал, подсказать не могу.

Это мультипекарь. Походу придется самому снимать)

alutov, 2022-11-14:
Пока у меня нет логов на это устройство. Будут логи, добавлю

Villi342, 2022-11-14:
Спасибо, постараюсь собрать на этой неделе

alutov, 2022-11-14:
Можно в телеграме, адрес есть на главной странице гита.

Villi342, 2022-11-14:
Хорошо

alutov, 2023-03-19:
Мультипекарь добавлен, но пока полностью не протестирован.

closed 2023-05-24 by alutov