alutov/ESP32-R4sGate-for-Redmond #10: Добавить поддержку чайника RK-M216S

acckii, 2020-10-18:
Можно ли добавить поддержку чайника RK-M216S. У меня не вышло его подключить, выбирал разные пункты меню. Может я что-то не так делаю или всё же RK-M216S не поддерживается?

Если всё же не поддерживается, то стоит ли ожидать поддержку RK-M216S и в какой срок? Если чем-то необходимо помочь, то я обязательно помогу.

alutov, 2020-10-18:
В новых версиях прошивок Redmond немного изменен ответ на авторизацию. Начиная с версии 2020.10.16 в шлюзе это учтено. Скачайте новую версию и пробуйте, Чайник должен присоединиться. Ну а выбрать можно 211 или 240.

acckii, 2020-10-19:
Всё отлично работает. Спасибо.

closed 2020-10-20 by alutov

QuAzI, 2024-01-04:
Кое-как один раз у меня увиделся чайник... и пропал... скажите пожалуйста, его что, нужно будет пэйрить каждый раз как я нажимаю сохранить настройки? Забэкапить/восстановить никак нельзя?

alutov, 2024-01-04:
Похоже, надо опять нажать кнопку включения для сопряжения чайника со шлюзом. Прочему пропала привязка, непонятно. У меня при сохранении настроек ничего не отваливается.