alutov/ESP32-R4sGate-for-Redmond #11: RK-G210S

Den-88, 2020-10-24:
Здравствуйте!
А возможно ли как-то добавить чайник RK-G210S?

alutov, 2020-10-25:
Думаю, будет работать. Выберите ближайший, например RK-G211S.

Den-88, 2020-10-25:
Пробовал все выбирать, но не подключается. Пишет, что последнее найденное устройство bluetooth - RK-G210S.

BertTomsk, 2020-10-25:
Всё работает, в настройках надо ещё имя чайника прописать - RK-G210S. Я единственное в новой прошивке не понял, как выключить мигающую подсветку, когда выключен чайник?

alutov, 2020-10-25:
А подсветка была включена из шлюза?

Den-88, 2020-10-25:
Я прописывал имя "Kettle" и не смог подключится. Сейчас на работе попробую с чайником 200 модели, а вечером ещё раз дома с 210 буду пробовать...

Den-88, 2020-10-25:
Вообщем к 200 подключился, но управлять ни подсветкой и чайником не хочет! В HA прокинулся без проблем. Управление ни из веб интерфейса, ни из  HA не работает! Только уровень сигнала показывает. И ещё точмозит веб интерфейс...
В чём может быть проблема???

alutov, 2020-10-25:
Тип чайника установлен в 200?

BertTomsk, 2020-10-25:
> А подсветка была включена из шлюза? Если включить и потом выключить подсветку из шлюза, все нормально выключается? Вроде недавно подсветку правил, вроде бы работает. Попробуйте самую последнюю версию.
Подсветка (ночник) работает нормально, включается и управляется через Web и в HA. Я про другую подсветку, которая плавно загорается и гаснет в цвет температуры воды, когда чайник выключен и стоит на подставке.

BertTomsk, 2020-10-25:
> Я прописывал имя "Kettle" и не смог подключится. Сейчас на работе попробую с чайником 200 модели, а вечером ещё раз дома с 210 буду пробовать...

У меня тоже rk-g210s, если выбрать 211 или 240, то всё работает.

Den-88, 2020-10-25:
Пришёл домой, всё настроил с 210, работает отлично! И интерфейс не тормозит! Видимо на работе что-то с сетью странное... Спасибо всем!
Ещё вопрос, а зачем настройки часового пояса? Или это для мультиварки?

alutov, 2020-10-26:
В последних версиях шлюза на чайник примерно раз в минуту отправляется время. Читал, что чайник с установленным временем как раз и позволяет оценивать температуру воды. Вот только где это отключить, я не знаю, думаю, что в родном приложении должна быть такая опция. 
Часовой пояс нужен для установки времени. в то же кофеварке, розетке и чайниках 211 и 240. Но, кроме того, шлюз может быть с экраном, на котором как раз часы, температура дома и на улице, а также на выходе котла, некоторые параметры устройств и снимок с камеры.

Den-88, 2020-10-26:
> отправляется время. Читал, что чайник с установленным временем как раз и позволяет оценивать температуру воды. Вот только где это отключить, я не знаю, думаю, что в родном приложении должна быть такая опция. Шлюз управлять

Ясно! Спасибо! Теперь управление чайником отчилно работает, а то мой RP4 до него с коридора через раз добивает своим блютузом))

DivanX10, 2020-10-26:
> Подсветка (ночник) работает нормально, включается и управляется через Web и в HA. Я про другую подсветку, которая плавно загорается и гаснет в цвет температуры воды, когда чайник выключен и стоит на подставке.

Мигающая подсветка, это, если включена функция **Световая индикация**. Зайдите в настройки родного приложения Redmon и выключите функцию **Световая индикация**, предварительно выключите шлюз, а после того как выключите эту функцию, включите шлюз и подсветка перестанет мигать.

BertTomsk, 2020-10-27:
> Мигающая подсветка, это, если включена функция **Световая индикация**. Зайдите в настройки родного приложения Redmon и выключите функцию **Световая индикация**, предварительно выключите шлюз, а после того как выключите эту функцию, включите шлюз и подсветка перестанет мигать.

Спасибо, помогло. 

Ещё заметил, что в родном приложении режим ночник светит в 2 раза ярче, чем через шлюз. Максимальная подсветка через шлюз это где-то половины яркости если управлять в родном приложении

closed 2020-10-27 by Den-88

alutov, 2020-10-27:
Добавил яркость подсветки в релиз 2020.10.27, нужно проверить.