alutov/ESP32-R4sGate-for-Redmond #12: Информация

DivanX10, 2020-10-26:
Было бы здорово, если при выходе новой прошивки было описание, что нового добавлено и что исправлено, чтобы было понимание, нужно ли обновляться. Также подробное описание лучше выкладывать в разделе Wiki. Например как [здесь пишут что изменилось в прошивке](https://github.com/Aircoookie/WLED/releases/tag/v0.10.2) и пример [Wiki](https://github.com/Aircoookie/WLED/wiki), а на главной странице лучше указать список поддерживаемых устройств. Составил список поддерживаемых устройств. Спасибо за отличную прошивку.

**Список поддерживаемых устройств**

**Электрочайник**
Redmond SkyKettle M171S
Redmond SkyKettle G200S
Redmond SkyKettle G210S
Redmond SkyKettle G211S
Redmond SkyKettle G212S
Redmond SkyKettle G216S
Redmond SkyKettle G240S

**Мультиварка**
Redmond SkyCooker RMC-M800S

**Кофеварка**
Redmond SkyCoffee M1519S

**Розетка**
Redmond SkyPort 103S

**Конвектор электрический**
Redmond SkyHeat 4529S

Den-88, 2020-10-27:
Поддерживаю!

alutov, 2020-10-27:
Ну да, перезалил где-то в 5 вечера релиз 2020.10.27, проверить нужно.

DivanX10, 2020-10-27:
> Ну да, перезалил где-то в 5 вечера релиз 2020.10.27, проверить нужно.
> Описание поправлю конечно, только следить за всеми моделями не успеваю, пусть и здесь висит информация. И да, у меня работает чайник 173.

Добавил ваш чайник в список

alutov, 2020-10-28:
Главная страница на русском есть. Где еще?

Jony26, 2020-12-03:
Доброго времени! Не могу понять чайник Redmond SkyKettle G214S поддерживается? В списке его нет...

alutov, 2020-12-04:
По идее должен. Пробуйте с 200 или 240, это одно и то же, протокол одинаковый.

NooBAsTiK, 2020-12-29:
А как подключить чайник RK-M216S? И второй вопрос как использовать плату по назначению с этим шлюзом? Мне нужно на одной ноге управлять PWM.

alutov, 2020-12-29:
Пробуйте тип чайника 200 или 240, это одно и то же, протокол одинаковый. Имя впишите из "last found device". Пока управление в шлюзе не предусмотрено. Заливаете свою прошивку и управляете.

NooBAsTiK, 2020-12-29:
200 поставил и получилось. Залить свою прошивку и шлюз одновременно можно?

alutov, 2020-12-29:
Нет, нужно выбирать)))

NooBAsTiK, 2020-12-29:
Жаль, придётся две esp ставить. Чайник работает подключил к openhab можете в список добавить Redmond SkyKettle RK-M216S

feanche, 2021-02-17:
Не знал куда еще написать, но думаю инфа пригодится. Для тех у кого термопот модели RTP-M810S, то выбирайте в настройках type: RK-M173S. Термопот успешно подключается к хабу и можно управлять режимами, смотреть температуру и прочее.

alidonet, 2021-03-04:
Поддерживает RK-G214S

dime21w, 2021-03-06:
> Поддерживает RK-G214S

Можете поподробнее что куда заполнить , а то мой 214S не работает.
[https://ibb.co/FhLjLgY](url)
[https://ibb.co/NLZkbGh](url)

DivanX10, 2021-03-06:
Просто перетащите картинку сюда 😉 

[image]

alidonet, 2021-03-06:
> Можете поподробнее что куда заполнить , а то мой 214S не работает.
[image]
[image]

AntoxaSa, 2021-06-24:
REDMOND SkyKettle G233S будет ли работать?

medjaiiii, 2021-07-05:
Присоединюсь, купил и жду REDMOND SkyKettle G202S, будет ли работать? Какой выбрать если нельзя добавить?

DivanX10, 2021-08-13:
> Приветствую! RK-G201S каким нибудь образом можно подключить?

Можно. Заходите в чат телеграм https://t.me/ESP32_R4sGate

vmorozov23, 2022-02-18:
здравствуйте, помогите настроить чайник  RK-G200S, установил на роутер mqtt брокер, купил esp32 прошил ее, esp видит чайник и подключена к mqtt брокеру, из esp32 чайник включается. но вот когда пытаюсь отправить команду на mqtt брокер, ничего не происходит, чайник не включается. 
[image]

alutov, 2022-02-18:
Ну да, описание чуть устарело, нужно писать в r4s/df16a0f981c3/cmd/boil или r4s/df16a0f981c3/cmd/heat для включения кипячения и подогрева соответственно. Или использовать совмещенные топики (common topics) в настройках, чтобы писать туда же, откуда и читать.

conrad66, 2022-06-13:
За что отвечает number.r4s01_kettle_boiltime  ?
Ни где не могу найти описание

vmorozov23, 2022-06-13:
Не знаю

alutov, 2022-06-13:
Время кипения чайника, значения от -5 до +5. Недавно добавлено. Описание mqtt давно не поправлял.

alutov, 2022-06-14:
Надежнее всего по переходу r4s01.Kettle.boil из вкл в выкл. Если чайник снимается с подставки будет переход в недоступно. Можно и по вычисленному уровню воды, но тут нужно, чтобы начальная температура при включении  кипячения была не выше 72  градусов. 75 - 3 градуса на включение.