alutov/ESP32-R4sGate-for-Redmond #13: Умная розетка Redmond RSP-103S

DivanX10, 2020-10-30:
Скажите, как я знаю, что данная розетка Redmond RSP-103S поддерживается шлюзом. 

1) Скажите, как она работает со шлюзом. Нет задержек? 
2) Управление такое же быстрое, как и с управлением чайником? 
3) Если подключить к шлюзу несколько устройств, шлюз не будет зависать? Не скажется на быстродействие в управлении чайником?

alutov, 2020-10-30:
У меня подключено 3 разных устройства. Вроде бы нормально все, пробуйте.

DivanX10, 2020-10-30:
Привезли розетку Redmond RSP-103S. Подключил. Работает без проблем.

DivanX10, 2020-10-31:
Обнаружился глюк в работе блокировки розетки. Если я включу блокировку розетки, то сама розетка не включается, но если я выключу блокировку, то розетка включается. Если включу розетку, а потом блокировку, то розетка выключится. Так не должно быть. Блокировка должна включаться не важно в каком режиме и она просто блокирует либо включение, либо выключение.

alutov, 2020-10-31:
Я думаю, как раз так и должно быть. Шлюз тут вообще никак не влияет. Он просто передает команду блокировки и все. Розетка выключается сама. В принципе, легко все проверяется в родном приложении.

DivanX10, 2020-10-31:
Оказалось все дело в самой прошивке данной розетки. Я когда распаковал розетку, то не стал изначально добавлять в родное приложение и зря. Когда добавил розетку в родное приложение, то программа предложила обновить розетку, что я и сделал. Сейчас у меня стоит прошивка версии 5.13. На старой прошивке у меня блокировка работала неправильно, но когда обновил розетку до прошивки версии 5.13, то блокировка стала работать правильно. Т.е если розетка выключена и я включу, а потом выключу блокировку, то розетка не включится, а если розетка включена и включить блокировку, то розетка выключится.

alutov, 2020-10-31:
А понял сейчас о чем именно пост. В наличии 2 розетки, одна была обновлена через приложение, другая нет. Так и есть, если розетка была включена, то после снятия блокировки старая версия включается, новая нет.

DivanX10, 2020-10-31:
Дожили, когда перепрошиваем розетку ). Никогда не мог об этом подумать, но прогресс идет. Выложил [видос](https://youtu.be/aUirgdTNhPo), как вариант применения данной розетки в быту.

closed 2020-11-21 by alutov