alutov/ESP32-R4sGate-for-Redmond #14: А нельзя ли доработать ваш шлюз для такого чайника ??

tarasfrompir, 2020-11-04:
Авторизация и все остальное описано тут 
https://github.com/aprosvetova/xiaomi-kettle

alutov, 2020-11-04:
Даже не знаю, что сказать... Теоретически можно, я смотрел бегло эту страничку, там чайник не включается, только поддерживается. Подумал тогда, что оно того не стоит.

tarasfrompir, 2020-11-04:
Там бы только функцию авторизации поправить, да функцию для отправки  команды - я уже и так и сяк а оно мне ни в какую... Человек просит - а я профан опростоволосился...
А чего не включается ? 

Хеатинг жеж есть вроде
[image]

DivanX10, 2020-11-05:
Главное, чтобы эту прошивку не сломали ). Все же она чисто для Redmond. Если будете добавлять чайник xiaomi в прошивку для Redmond, то лучше сделать отдельную прошивку для Xiaomi.

tarasfrompir, 2020-11-05:
Да я ж не против - только за

alutov, 2020-11-08:
Разве уже нет готовых решений? Чтобы это сделать, надо чайник покупать как минимум.

tarasfrompir, 2020-11-08:
> Разве уже нет готовых решений? Чтобы это сделать, надо чайник покупать как минимум.

если бы было - не просил бы - или ткните носом....

alutov, 2020-11-08:
Обещать ничего не буду. Чайник заказал.

alutov, 2020-11-20:
Добавил чайник MiKettle, можно пробовать.

tarasfrompir, 2020-11-20:
> Добавил чайник MiKettle, можно пробовать.

Пасибулечки огромное

tarasfrompir, 2020-11-20:
есть в коннекте мажордомо регистрация ?

alutov, 2020-11-20:
Нет регистрации, оно мне не нужно.

tarasfrompir, 2020-11-20:
> Нет регистрации, оно мне не нужно.

Да там у меня немного русских денег есть - больше пока благодарить нечем

alutov, 2020-11-20:
В принципе не нужно.

tarasfrompir, 2020-12-11:
А можно пример команды через ВЕб - любой

alutov, 2020-12-11:
Не понял вопрос, зачем? Чайник подсоединили к шлюзу? Если да, то при входе на страничку управления он должен написать "Press "Warm" on MiKettle to remote device control". Нажимаете сенсор "Warm" на чайнике и открывается меню управления. Управлять можно только из этого режима, в доке это отмечено, как впрочем и в вебе.

tarasfrompir, 2020-12-11:
Я прошу прощения мои выражовувания не верны
вот это  можно описать в текущей реализации ? или только МКЮТТ ?
HTTP:
http://r4sgate.local/r4s/[devid]/off - выкл.
http://r4sgate.local/r4s/[devid]/heat?temp=60 - нагрев
http://r4sgate.local/r4s/[devid]/boil - кипячение
http://r4sgate.local/r4s/[devid]/boil?temp=95 - кипячение и нагрев

alutov, 2020-12-11:
Http шлюз не поддерживал никогда.  Либо web, либо Mqtt.

WildeRNS, 2020-12-23:
Спасибо за отличную и главное рабочую прошивку для чайника Xiaomi Mi Kettle!
Есть пара вопросов: 
1. После подключения напрочь пропадает возможность работы чайника из MiHome (чтобы возобновить работу, приходится удалять чайник и заново добавлять), так задумано?
2. Нельзя ли добавить галочку retain,чтобы при снятии/установки чайника с бази, предыдущая температура не сбрасывалась на 0? (Использую Home Assistant)

alutov, 2020-12-23:
1. Нельзя пока что. Шлюз при подключении использует свой ключ, а не берет из михома. И использует полную процедуру. Смарт полную процедуру использует только при добавлении. Чайник похоже запоминает одно соединение.
2. Поправлено уже, это привет от совмещенных топиков. Если использовать раздельные, то и на текущей версии дефекта не будет.
3. По поводу http при написании не сильно разбирался, что там реально ходит, а сейчас нет времени. Есть проблемы с бле на есп32, в описании это есть, копаюсь с этим, пытаюсь локализовать.

tarasfrompir, 2020-12-24:
> 1. Нельзя пока что. Шлюз при подключении использует свой ключ, а не берет из михома. И использует полную процедуру авторизации при каждом соединении. Смарт полную процедуру использует только при добавлении. Чайник похоже запоминает одно соединение при работе по упрощенной процедуре авторизации.
> 2. Поправлено уже, это баг-привет от совмещенных топиков. Если использовать раздельные, то и на текущей версии дефекта не будет.
> 3. По поводу http при написании не сильно разбирался, что там реально ходит, а сейчас нет времени. Есть проблемы с бле на есп32, в описании это есть, копаюсь с этим, пытаюсь локализовать.

Решил через curl - вроде работает

ZL0V0N, 2021-07-26:
Добрый день. Можно добавить поддержку Redmond G203S ?

closed 2021-12-10 by alutov