alutov/ESP32-R4sGate-for-Redmond #16: Добавить отключение\включение дисплея по MQTT

Saniksanik, 2020-11-21:
Добавить топик, по которому бы включался или отключался экран

alutov, 2020-11-21:
Подскажите, это какой регион, как его назвать?

alutov, 2020-11-21:
Неужели есть кто-то , кто припаял экран?

Saniksanik, 2020-11-21:
Я использую готовый модуль TTGO T4 BTC Ticker ESP32.
Он удобен тем, что есть корпус. Естественно, пины пришлось поправить в прошивке.
Суть просьбы - может быть можно как-то отключать дисплей по внешней команде? К примеру, читать некий mqtt-topic и если там приходит off - выключать дисплей. Ночью не очень удобно что он светится.

alutov, 2020-11-22:
Я как-то о готовых модулях не подумал. А раз так, нужно еще сделать выбор пинов. Не пересобирать же каждый раз прошивку. Буду делать.

alutov, 2020-11-23:
Подскажите пожалуйста, какие пины куда идут в модуле TTGO T4 BTC Ticker ESP32.  Не нахожу ограничений на установку, а ставлю, например CLK 20 и esp виснет, ругаясь на неверный пин.

Saniksanik, 2020-11-23:
Вообще на этих модулях разные пины и в принципе их нормально задавать при компиляции. Пины зависят от ревизии платы и только я знаю 3 таких ревизии.
Для примера моя ревизия 1.3 имеет такие пины
#define TFT_BL      4

#define SD_MISO     2
#define SD_MOSI     15
#define SD_SCLK     14
#define SD_CS       13

#define I2C_SDA     21
#define I2C_SCL     22

#define SPEAKER_PWD 19
#define SPEAKER_OUT 25
#define ADC_IN      35

#define BUTTON_1    37
#define BUTTON_2    38
#define BUTTON_3    39
#define BUTTON_4    0

По поводу дисплея. В других библиотеках для работы с данным дисплеем еще был угол поворота экрана - здесь не нашел, а хотелось бы повернуть изображение на 180 градусов. Если такое есть - подскажите. Если нет - буду колхозить другой вариант подвода питания. а не от microUSB

Saniksanik, 2020-11-24:
Огромное спасибо, на выходных протестирую

Saniksanik, 2020-11-24:
Также в список рекомендуемого оборудования можете добавить m5 stack basic, я проверил прошивку на этом модуле.
Он, конечно избыточен и немного дороговат, но зато имеет красивый корпус и его проще купить чем мой китайский BTC Ticker.
Ссылка https://m5stack.com/products/basic-core-iot-development-kit
Для тех кто не хочет сам возиться с корпусом, а хочет просто залить прошивку и получить готовое устройство - идеальный вариант

vorlocsev, 2021-05-01:
есть ли поддержка дисплеев на том же чипе но 480×320, пробовал менять стандартные пины тоже не взлетело.

alutov, 2021-10-18:
Купил для пробы m5stack core, но не могу запустить экран. Не подскажете, на каких пинах он поднялся?

closed 2021-12-10 by alutov