alutov/ESP32-R4sGate-for-Redmond #17: Bluetooth шлюз для меток\телефонов

DivanX10, 2020-11-25:
Можете сделать отдельную прошивку Bluetooth шлюз для меток\телефонов по аналогии шлюза ESP32-R4sGate-for-Redmond? Все тоже самое, только разница в том, что вместо устройств Redmond будут добавляться bluetooth метки или телефоны. В веб интерфейсе к примеру будет отображаться список мак адресов устройств и то устройство, которое нужно зафиксировать, добавляем в список обнаружения и оповещения о доступности\не доступности. Допустим я дома, шлюз увидел телефон и передал значение 1 в УД, если меня нет дома, то шлюз передает значение 0. Тоже самое это работает и с чайником в шлюзе Redmond. Есть конечно OpenMQTTGateway, но это совершенно не то и даже близкого такого функционала нет.

alutov, 2020-12-12:
Может быть, но пока не думал об этом.

valek1011, 2021-03-07:
Это было бы очень супер:-)

DivanX10, 2021-03-08:
На самом деле это лишнее, можно сделать другим способом.

1) Если стоит MajorDoMo, то скачиваем прошивку [отсюда](https://github.com/shogunxam/ESP32_BLETracker) и заливаем на ESP32, далее подключаем ESP по MQTT к MajorDoMo и указываем пути к свойствам. Аналогично со шлюзом r4s при  подключении чайника.

2) Если стоит Home Assistant, то ставим аддон ESP Home и вставляем код

```
esphome:
  name: esp_ble_tracker
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "имя wi-fi"
  password: "пароль"

  # Точка доступа. Если в течение минуты модуль не сможет подключиться к сети, то должна подняться точка доступа.
  ap:
    ssid: "придумываем имя для точки доступа"
    password: "придумываем свой пароль"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

# Кнопка перезапуска    
switch:
  - platform: restart
    name: ble_esp32_Restart

# Конфигурация для BLE трекера
esp32_ble_tracker:

# Состояние устройства. Указываем нужное количество трекеров, ниже я указал 3 трекера, если нужно больше, то просто # копируем и вставляем ниже
binary_sensor:
# Трекер iTag: "любое имя\название"
  - platform: ble_presence
    mac_address: 00:00:00:00:00:00 # указываем тут свой mac адрес трекера
    name: "user1" # вместо user указываем имя владельца трекера
    
# Трекер iTag: "любое имя\название"
  - platform: ble_presence
    mac_address: 00:00:00:00:00:00 # указываем тут свой mac адрес трекера
    name: "user2" # вместо user указываем имя владельца трекера

# Трекер iTag: "любое имя\название"
  - platform: ble_presence
    mac_address: 00:00:00:00:00:00 # указываем тут свой mac адрес трекера
    name: "user3" # вместо user указываем имя владельца трекера    

# Сенсоры. Размещаем сюда все устройства с которых нам нужно получить информацию  
sensor:
# Уровень сигнала Wi-fi
  - platform: wifi_signal
    name:  ble_esp32_wifi
    update_interval: 60s

# Датчик времени безотказной работы позволяет отслеживать время работы ESP в течение нескольких секунд.
  - platform: uptime
    name: Uptime Sensor

# Указываем нужное количество трекеров, ниже я указал 3 трекера, если нужно больше. то просто копируем и   
# вставляем ниже

# Трекер iTag: user1 # вместо user указываем имя владельца трекера для понимания чей этот трекер
  - platform: ble_rssi
    mac_address: 00:00:00:00:00:00
    name: "ESP32 BLE user1 RSSI value" # вместо user указываем имя владельца трекера

# Трекер iTag: user2 # вместо user указываем имя владельца трекера для понимания чей этот трекер
  - platform: ble_rssi
    mac_address: 00:00:00:00:00:00
    name: "ESP32 BLE user1 RSSI value"

# Трекер iTag: user3 # вместо user указываем имя владельца трекера для понимания чей этот трекер
  - platform: ble_rssi
    mac_address: 00:00:00:00:00:00
    name: "ESP32 BLE user3 RSSI value"
  
``` 
Так выглядит в Home Assistant
[image]

alutov, 2022-03-31:
Добавил эту функцию в прошивку шлюза.

closed 2022-04-30 by alutov