alutov/ESP32-R4sGate-for-Redmond #18: ESP32 не подключается к роутеру, но подключается к телефону

balbulator, 2020-12-01:
Здравствуйте.
После прошивки ESP32 не подключается к роутеру по Wi-Fi к сети r4s c паролем 12345678, это прослеживается в SerialMonito. 
Но если создать точно такую же точку доступа на смартфоне, то подключается, правда в непонятной мне подсети 192.168.43.14, хотя моя локалка 192.168.0.1 - 192.168.0.254. В телефоне могу зайти в веб интерфейс, но с MQTT брокером связь установить уже не могу, наверное моя подсеть недоступна. В интернете где-то уже видел такую проблему, но решения не там не было.
Win7 x64, роутер D-Link DIR-815, ESP32 DEVKITV1
Если есть возможность, помогите разобраться, а то чаю давно уже не пил)))
Спасибо

alutov, 2020-12-01:
Если в телефоне заходите в веб, то установите логин и пароль домашней сетки Wifi. Когда есп подключится к ней, установите остальное.

balbulator, 2020-12-01:
Так и сделал, но к домашней сети ESP32 так и не подключается. Не могу понять какая разница. Единственная настройка в телефоне точки доступа это - защита WPA2-Personal. Думаю на роутере это соответствует WPA2-PSK. Пробовал различные сетевые аутентификации, но увы. Чайник думаю пока что не причем Xiaomi Kettle.
Какие еще есть догадки?

alutov, 2020-12-02:
Была проблема в подключении, но там в имени сети были не английские буквы и не цифры, в частности запятая.  Они особо кодируются при передаче параметров в вебе. Тогда поправил, но возможно не полностью. Но почему не подключается тогда к сети с именем r4s и паролем 12345678 непонятно. WPA2-PSK должна поддерживаться, но можно попробовать в роутере и просто WPA-PSK. Может есп попала в черный список роутера? Ну и чтобы исключить имя с паролем, если на телефоне сделать точку доступа с параметрами из роутера, к ней подключается?

balbulator, 2020-12-02:
Все синтаксисы в написании логина и пароля правильны, ибо не смогли бы подключиться другие ESP и Raspberry. Пробовал создать точку доступа на телефоне с таким же именем и паролем, к телефону подключается без проблем, даже несмотря на то что роутер с таким же именем и паролем даже не отключал. В роутере пробовал и в гостевом режиме и в основном и с разными логинами и режимами WPA-PSK. Это какой-то нонсенс. Так как у меня умный дом, а это порядка 30 клиентов только по Wi-Fi и в том числе ESP32, нет абсолютно никаких проблем с подключением. Проверю также идею с расстоянием до роутера и другой прошивкой, чтобы исключить брак самой платы.
За желание помочь - благодарю, выслушаю все советы

balbulator, 2020-12-02:
Проверил плату - рабочая, сигнал пробивает через всю квартиру (отбросим проблему железа)
Я не слишком разбираюсь в логах, но на всяк случай оставлю здесь

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7016
load:0x40078000,len:14440
load:0x40080400,len:3980
entry 0x40080680
[0;32mI (29) boot: ESP-IDF v4.2-dev-1905-g625bd5eb1 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 05:39:10[0m
[0;32mI (29) boot: chip revision: 1[0m
[0;32mI (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (40) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (45) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (49) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (54) boot: Enabling RNG early entropy source...[0m
[0;32mI (59) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (77) boot:  1 phy_init         RF data          01 01 0000d000 00001000[0m
[0;32mI (85) boot:  2 otadata          OTA data         01 00 0000e000 00002000[0m
[0;32mI (92) boot:  3 ota_0            OTA app          00 10 00010000 001e0000[0m
[0;32mI (100) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000[0m
[0;32mI (107) boot:  5 spiffs           Unknown data     01 82 003d0000 00030000[0m
[0;32mI (115) boot: End of partition table[0m
[0;32mI (119) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (126) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x52720 (337696) map[0m
[0;32mI (263) esp_image: segment 1: paddr=0x00062748 vaddr=0x3ffbdb60 size=0x055ec ( 21996) load[0m
[0;32mI (272) esp_image: segment 2: paddr=0x00067d3c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (273) esp_image: segment 3: paddr=0x00068144 vaddr=0x40080400 size=0x07ed4 ( 32468) load[0m
[0;32mI (295) esp_image: segment 4: paddr=0x00070020 vaddr=0x400d0020 size=0x11127c (1118844) map[0m
[0;32mI (718) esp_image: segment 5: paddr=0x001812a4 vaddr=0x400882d4 size=0x15cd4 ( 89300) load[0m
[0;32mI (773) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (773) boot: Disabling RNG early entropy source...[0m
[0;32mI (773) cpu_start: Pro cpu up.[0m
[0;32mI (777) cpu_start: Starting app cpu, entry point is 0x40081108[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (799) cpu_start: Pro cpu start user code[0m
[0;32mI (799) cpu_start: cpu freq: 160000000[0m
[0;32mI (799) cpu_start: Application information:[0m
[0;32mI (803) cpu_start: Project name:     r4sGate[0m
[0;32mI (808) cpu_start: App version:      1.0[0m
[0;32mI (813) cpu_start: Compile time:     Nov 29 2020 18:46:38[0m
[0;32mI (819) cpu_start: ELF file SHA256:  dcc05b3bcdaa60e4...[0m
[0;32mI (825) cpu_start: ESP-IDF:          v4.3-dev-1901-g178b122c1-dirty[0m
[0;32mI (832) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (839) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (845) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (851) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (857) heap_init: At 3FFD2FF0 len 0000D010 (52 KiB): DRAM[0m
[0;32mI (863) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (870) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (876) heap_init: At 4009DFA8 len 00002058 (8 KiB): IRAM[0m
[0;32mI (883) spi_flash: detected chip: generic[0m
[0;32mI (887) spi_flash: flash io: dio[0m
[0;32mI (892) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
Starting r4sGate...
This is esp32 chip with 2 CPU cores, WiFi/BT/BLE, 
silicon revision 1, 4MB external flash
[0;32mI (921) R4S: Init start free heap: 181396
[0m
APP IDF version: v4.3-dev-1901-g178b122c1-dirty
[0;32mI (2211) R4S: ILI9341 TFT not found.[0m
[... 75 lines cut ...]
[0;32mI (27611) R4S: retry to connect to the AP[0m
[0;32mI (27611) R4S: connect to the AP fail[0m
I (27851) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (27861) wifi:state: init -> auth (b0)
I (28861) wifi:state: auth -> init (200)
I (28861) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
[0;32mI (28861) R4S: connect to the AP fail[0m
[0;32mI (28861) R4S: Failed to connect to SSID:'mylogin', password:'mypassword'[0m
I (28871) wifi:flush txq
I (28871) wifi:stop sw txq
I (28871) wifi:lmac stop hw txq
I (28911) wifi:mode : sta (c4:4f:33:75:7e:d5)
I (28911) wifi:enable tsf
[0;32mI (28911) R4S: Connecting to ap SSID:'r4s' password:'12345678'[0m
I (32041) wifi:new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
I (32941) wifi:state: init -> auth (b0)
I (33941) wifi:state: auth -> init (200)
I (33941) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
[0;32mI (33951) R4S: connect to the AP fail[0m
[0;32mI (33951) R4S: Failed to connect to SSID:'r4s', password:'12345678'[0m
[0;32mI (33951) R4S: Restarting now...[0m
I (33961) wifi:flush txq
I (33961) wifi:stop sw txq
I (33961) wifi:lmac stop hw txq
ets Jun  8 2016 00:22:57`

alutov, 2020-12-03:
Попробовал вчера менять режимы на роутере. У меня есп нормально подключается  с WPA_PSK, WPA2_PSK, WPA3_PSK.
По логу вроде как не проходит авторизацию.

balbulator, 2020-12-04:
Я тут поэкспериментировал с прошивками лампы GyverLamp. Все подключилось, хотя в прошивке нашел пару глюков, но в остальном все подключилось к сети на ура (роутер помилован). Правда прошивка там с открытым кодом и есть возможность покопаться (в Ардуино). А на какой именно плате у Вас реализовано, возможно несовместимость какая, у меня Wi-Fi ESP-Wroom-32, плата ESP32 DEVKITV1. Премного благодарен за содействия в поисках истины.

alutov, 2020-12-05:
Нет авторизации, это программное что-то, не понимаю, причем тут плата. Может в черный спискок есп32 попала. ну или пароль.  Может его вводите и правильно, а есп32 его по-другому понимает. Но как я понимаю, в логе нужный пароль. Вы роутер перегружали? 
А код тут открытый, и [кто хочет, тот правит](https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/16).

balbulator, 2020-12-07:
Подменил роутер на Huawei WS319, разместил очень близко, отключил все фильтра по МАК и АйПи. Результат все тот же. Создавал гостевую сеть r4s, перезагружал, все без результата

```
`[0;32mI (35040) R4S: Connecting to ap SSID:'r4s' password:'12345678'[0m
I (38180) wifi:new:<13,2>, old:<1,0>, ap:<255,255>, sta:<13,2>, prof:1
I (39100) wifi:state: init -> auth (b0)
I (40100) wifi:state: auth -> init (200)
I (40100) wifi:new:<13,0>, old:<13,2>, ap:<255,255>, sta:<13,2>, prof:1
[0;32mI (40110) R4S: connect to the AP fail[0m
[0;32mI (40110) R4S: Failed to connect to SSID:'r4s', password:'12345678'[0m
[0;32mI (40110) R4S: Restarting now...[0m
I (40120) wifi:flush txq
I (40120) wifi:stop sw txq
I (40120) wifi:lmac stop hw txq
ets Jun  8 2016 00:22:57`
```
Завтра попробую поиграться с каналами, больше версий у меня нет

alutov, 2020-12-07:
У меня тоже. В переписке информации ноль.  В роутерах вообще-то есть журнал, где можно посмотреть по какой именно причине идет отклонение соединения. Попробуйте есп32, Или поищите другой шлюз.

closed 2020-12-07 by alutov

balbulator, 2020-12-07:
Вот нашел аналогичную проблему https://community.alexgyver.ru/threads/esp-32-problemy-podkljuchenija.1545/