alutov/ESP32-R4sGate-for-Redmond #19: REDMOND SkyTracker 08S

DivanX10, 2020-12-05:
Получится ли добавить данный трекер в прошивку? Данный трекер обнаруживается ESP32.
https://redmond.company/ru/products/sky-home/umnyy-treker-redmond-skytracker-08s/

DivanX10, 2020-12-06:
Нашел отличную [прошивку](https://github.com/shogunxam/ESP32_BLETracker) для работы с трекером REDMOND SkyTracker 08S. Данный трекер REDMOND SkyTracker 08S прекрасно работает с этой прошивкой.
[image]

Получится ли это добавить в шлюз Redmond?

alutov, 2020-12-12:
Если есть готовые решения, то пока не в приоритете. Ну может быть, если трекер мне понравится.

ViktuK359, 2021-04-23:
Вот было бы отлично, добавить в прошивку штатную функцию ESPhome как ESP32 Bluetooth Low Energy Tracker Hub ))

alutov, 2022-05-11:
REDMOND SkyTracker 08S виден в BLE мониторе.

closed 2022-05-11 by alutov