alutov/ESP32-R4sGate-for-Redmond #21: Поддержка мультиварки RMC-M92S ?

sisaenkov, 2020-12-22:
Есть возможность поддержки мультиварки RMC-M92S?
В чём может быть сложность ее добавления?

alutov, 2020-12-24:
Нужно пробовать. Вписываете свое имя в device name и тип выбираете rmc80. Что-то должно работать, программы, наверное не совпадут. Но в любом случае, пока не будет информации, добавить ничего не смогу.

Yevgen16, 2021-01-02:
Для проверки своей мультиварки RMC-M42S тоже подключил шлюз. Действительно, программы не совпадают. Это, наверное, можно поправить. И ещё, моя мультиварка, сразу не включается после установки программы. Ждет типа нажатия кнопки старт. А так как нажатия через шлюз не происходит, то где-то через минуту программа сбрасывается и мультиварка становится в режим ожидание. Можно ли как-то это дело исправить? Готов проделать нужные движения :-)

alutov, 2021-01-02:
Нажатия через шлюз не происходит, потому что не нажимаете. Если мультиварка выключена, то по умолчанию шлюз ждет установки программы, при этом в "Select state" выбрано "Set Program". После выбора режима из меню "Select program" шлюз ждет корректировки при необходимости параметров по умолчанию. Если все устраивает, то в "Select state" нужно выбрать "Start Program". Да, если программы не совпадают - это неудобно, но не критично. В mqtt программы устанавливаются цифрами, и проблема может быть только в том, что если программ больше 12, то часть из них будет нельзя выставить.

Yevgen16, 2021-01-02:
"Если все устраивает, то в "Select state" нужно выбрать "Start Program"."... Хм, а у меня в "Select state" что-то не видно "Start Program". Только три пункта: "Switch off", "Warming on", "Set program".

alutov, 2021-01-02:
Еще раз пишу: "После выбора режима из меню "Select program" шлюз ждет корректировки при необходимости параметров по умолчанию. Если все устраивает, то в "Select state" нужно выбрать "Start Program". ПОСЛЕ ВЫБОРА ПРОГРАММЫ!!!

Yevgen16, 2021-01-02:
Ну да. Программа отправилась в мультиварку. Время и температура появились. Но в поле "Select state" "Start Program"  не могу выбрать, нет его :-(
[image]

alutov, 2021-01-02:
Ага, не только программы не совпадают. Не совпадает ответ на запрос статуса. И потому после нажатия ок температура и время не появляются в шлюзе. Нули так и остались. Ну тогда только лог снимать.

Yevgen16, 2021-01-02:
Понял. Значит с наскоку не получится. А как лог снять? Если подскажите, то сделаю. Если вам, конечно, интересно.

alutov, 2021-01-02:
[Здесь ](https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/6) данные по 800 мультиварке, там и список программ. В папку build положил файл r4sGatex.bin. В нем отключил проверку длины статуса. Может что-то появится. Если нет, нужно соединять смартфон в мультиваркой, разрешить в параметрах разработчика вывод лога блютуса в файл, давать по очереди все команды, а потом сливать лог со смартфона и анализировать его.

Yevgen16, 2021-01-02:
Понял. На днях займусь. Постараюсь сделать.

alutov, 2021-01-02:
Файл можно проверить быстро. И еще можно просто снять лог с самого шлюза при подаче команд. Там скорость 11520 8бит 1стоп без четности. Можно определиться хотя бы с длиной статусного сообщения.

Yevgen16, 2021-01-02:
r4sGatex.bin в папке build не вижу чего-то

alutov, 2021-01-02:
[Вот](https://github.com/alutov/ESP32-R4sGate-for-Redmond/blob/master/build/r4sGatex.bin)

Yevgen16, 2021-01-02:
Через веб обновил, ничего не изменилось. В мультиварку отправляется, но запуска программы не происходит.

alutov, 2021-01-02:
Ну тогда лог шлюза смотрите.

Yevgen16, 2021-01-02:
Не-а, не появляются, также нули. Да, сбрасывается программа.

MrC0nst, 2021-01-16:
Всем привет. На какой стадии завис проект по добавлению M92S?

alutov, 2021-01-16:
Проект и не начинался. Нет информации, кроме факта, что есть различия в протоколе 800 и 92.

alutov, 2021-02-28:
Добавлена в проект мультиварка RMC-903S. Отличается от RMC-800S строка состояния. Можно попробовать, приходит ли в шлюз состояние 92 и 42 мультиварки в режиме RMC-903S.

vsergeenko777, 2021-12-18:
Являясь обладателем RMC-M92S попробовал привязать её к шлюзу как RMC-903S. Работает довольно неплохо, есть возможность выбора и старта программы, включение/выключение автоподогрева после готовки, управления температурой и временем, в том числе отложенный старт. Список программ полностью соответствует доступным, даже экспресс работает отлично.

Обнаруженные проблемы:
1. Состояния не соответствуют действительности - например при готовке шлюз распознает состояние как DelayedStart, а подогрев WaitPasta. Экспериментальным путем подобрал таблицу истинных состояний для этой мультиварки:
```
1 = Setting
2 = DelayedStart
3 = кипячение воды для макарон, запускается при Start Program с выбранной программой Pasta (это WaitPasta?)
4 = ожидание макарон после кипячения воды (или это WaitPasta?)
5 = On (сам процесс приготовления, если я правильно понимаю)
6 = Warming
```
2. Попытка включить разогрев выбрав Warming On в Select state ничего не дает. Просто ничего не происходит. Возможно, как-то связано с состояниями. Остальные команды типа Switch Off, Set Program и Start Program отрабатывают штатно.

alutov, 2021-12-19:
Программы все совпадают один в один с 903? Лишнего нет в 903?

vsergeenko777, 2021-12-19:
Нет, ничего лишнего, всё совпадает и работает. Перепроверил каждую программу.

alutov, 2021-12-21:
Нужно пару дней, потом добавлю 92. Возможно, нужно будет снять лог родного приложения для получения команды подогрева. А как можно попасть в каждое состояние Wait_pasta?

alutov, 2021-12-21:
В папку jpg  положил бинарник для тестирования.

vsergeenko777, 2021-12-21:
Спасибо, скоро проверю.

Оба состояния WaitPasta это два этапа приготовления макарон, сначала надо запустить программу "макароны", мультиварка будет кипятить воду в состоянии 3 заместо 5 как в других программах, после кипячения (с звуковым сигналом) переходит в состояние 4 ожидая макароны в чаше. После добавления макарон нажимаем старт и состояние переключается в 5 как обычно при приготовлении.

Родное приложение покурю, отпишу если добуду какую-нибудь информацию.

vsergeenko777, 2021-12-21:
Проверил тестовую прошивку. Состояния теперь распознаются правильно (в том числе Boil4Pasta и WaitPasta), весь прежний функционал тоже работает отлично. По прежнему не работает включение разогрева. Кстати, при его включении вручную отображается state=6 и prog=17, а в приложении разогрев находится в конце списка программ. Но, как я понимаю, шлюз так же пытается включить его установкой 17-той программы и в случае с M92S это не работает. Попытаюсь посмотреть, как это делает родное приложение.

Заметил, что в Home Assistant перестали видеться объекты мультиварки, но как я понял это временно и будет исправлено в релизной версии.

vsergeenko777, 2021-12-21:
Проверил тестовую прошивку. Состояния теперь распознаются правильно (в том числе Boil4Pasta и WaitPasta), весь прежний функционал тоже работает отлично. По прежнему не работает включение разогрева. Кстати, при его включении вручную отображается state=6 и prog=17, а в приложении разогрев находится в конце списка программ. Но, как я понимаю, шлюз так же пытается включить его установкой 17-той программы и в случае с M92S это не работает. Попытаюсь посмотреть, как это делает родное приложение.

alutov, 2021-12-22:
Вечером постараюсь сделать вариант. За Ассистент спасибо, просмотрел, поправлю. Во предыдущих мультиварках разогрев включался без выбора программы. А в 92 они разогрев выделили в отдельную программу. Интересуют остальные параметры, какая температура и какое время идет по умолчанию.

vsergeenko777, 2021-12-22:
В HASS мультиварка всё же появилась через какое-то время, поэтому я стер эту часть из своего прошлого коммента :)

Json String через две минуты после включения разогрева:
```
{"mqtt":1,"c3503c6f8f2b":{"name":"RMC-M92S","status":"online","state":{"prog":17,"mode":0,"temp":70,"sethours":0,"setmins":30,"state":6,"hours":0,"mins":2,"warm":1}}
```
Температура 70°C, а вот времени статичного у программы разогрева нет, при его включении просто начинается отсчет прошедшего времени.

vsergeenko777, 2021-12-22:
Проверил, теперь работает. Сначала Set Program с программой Warming, затем Start Program и разогрев включается как вручную :)

Состояние при этом как и должно быть:
```
{"mqtt":1,"c3503c6f8f2b":{"name":"RMC-M92S","status":"online","state":{"prog":17,"mode":0,"temp":70,"sethours":0,"setmins":30,"state":6,"hours":0,"mins":6,"warm":1}}
```

Плюсом ещё раз перепроверил весь функционал и при управлении вручную и шлюзом - всё так же хорошо работает. Мультиварка RMC-M92S теперь полностью поддерживается, спасибо :)

closed 2021-12-23 by alutov