alutov/ESP32-R4sGate-for-Redmond #26: Отключается Idle режим в Mi Kettle

dmatora, 2021-01-13:
Возможно я так и не понял как правильно пользоваться R4SGate c чайником Xiaomi MiKettle YM-K1501(HK) (прошивка последняя - 6.2.1.9)
Пробовал 3 варианта
1. Оставлять чайник к режиме Idle (heat_temp = 0)
2. Отправлять чайник в режим Idle и отключать питание
3. Отправлять чайник в режим Heat и отключать питание

Во всех трех случаях чайник управляем (в случае 1 управляем после выключения/включения) только если включать чайник пока жидкость теплая (при остывании с 40 градусов до 20 это максимум 3 часа). Если оставить чайник на ночь, к утру во всех сценариях управляемость исчезает (при подаче питания, режим подогрева не включается)

Даже если включить подогрев сенсором "warm" на чайнике, отправить чайник в Idle и выключить питание до того как режим подогрева успел нагреть жидкость, при подаче питания режим подогрева не включается.

У кого-то вообще управление чайником Xiaomi при остывшей жидкости работает?

alutov, 2021-01-13:
Для начала спасибо за интересный вопрос в отличие [от](https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/24) и [от](https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/18). Для начала, возможно у Вас старая версия, лучше обновиться до 2021.01.10. Во-первых, был баг - при установке на подставку шлюз выключал чайник, если использовались совмещенные топики. Во-вторых, в уже скомпилированной библиотеке, то есть закрытой части esp-idf  есть [проблема зависания ble соединений](https://github.com/espressif/esp-idf/issues/6073), которая проявляется в той или иной мере во всех версиях при определенных условиях (возможно помехи по wifi и/или ble). К сожалению  никто не указал здесь на ее существование, обнаружил сам случайно. Хотя на сторонних ресурсах о ней упоминалось. Решится она не скоро, а пока добавил мониторинг ble. Сейчас вроде бы работает более-менее сносно.  У меня вроде бы чайник стоит сутками режиме Idle и при снятии и установке переходит в режим "warm"(при условии, что  этот режим до этого не был выключен кнопкой, тогда да, рестарт не помогает). Если при включении чайника лампа "warm" загорается, а управления нет, это как раз висит ble. При этом  по идее шлюз должен показывать чайник в offline. Сам режим Idle включается периодической посылкой команды установки таймера на 0 и установкой текущей температуры для более быстрого выполнения команды. В принципе команда полезна при возможности включения режима "warm", которую пока не нашел. Если использовать розетку, то она в принципе не нужна. Но и не вижу особой проблемы оставлять чайнике в режиме "warm". В этом режиме (по крайней мере у меня) чайник не включен постоянно, а поддерживает заданную шлюзом минимальную температуру 40°C с гистерезисом примерно 4°C, то есть при 36°C подогрев включается, а при 44°C отключается. И да, режим "warm" при старте чайника, а также таймер режима "warm" устанавливается на 12 часов и каждый раз при соединении. После 700 минут шлюз пытается сбросить таймер кратковременно включая и сразу же выключая кипячение. К сожалению, другого способа сброса таймера не нашел. Вообще, все это находится за рамками стандартного управления и может не срабатывать в новых версиях прошивки.

dmatora, 2021-01-13:
warm плох как минимум тем что я просыпаюсь от любого шороха, а уж от схлопывающихся пузырьков в воде - тем более.

Странно что версия прошивки чайника одинаковая, а ведет себя у меня и у Вас по разному. А Вы проверяли температуру? Если на ночь вскипятить, то к утру вода и не успеет остыть.

С проблемой зависания ble соединений я сталкивался, но быстро решил ее добавив перезагрузку шлюза перед ключевыми командами.
```
kettle_wake:
  icon: "mdi:kettle"
  alias: "Разбудить из подогрева"
  sequence:
    - service: shell_command.r4s_restart
    - service: switch.turn_on
      data:
        entity_id: switch.koogeek_p1_77093e
    - wait_template: "{% if states('climate.r4s1_kettle_temp') == 'off' and state_attr('climate.r4s1_kettle_temp', 'current_temperature') == 0 %}true{% else %}false{% endif %}"
    - wait_template: "{% if states('climate.r4s1_kettle_temp') == 'off' and state_attr('climate.r4s1_kettle_temp', 'current_temperature') > 0 %}true{% else %}false{% endif %}"
    - service: mqtt.publish
      data:
        topic: 'r4s/087cbe8b1826/cmd/heat_temp'
        payload: '40'
```
Периодическая установка сна не помогает, пробовал выводить из сна раз в час и отправлять туда снова через автоматизацию в Home Assistant - как только вода остывает, выход из сна перестает работать.

Может я не правильно отправляю в сон? Но тогда не ясно почему он работает первые три часа
```
    - service: mqtt.publish
      data:
        topic: 'r4s/087cbe8b1826/cmd/heat_temp'
        payload: '0'
```

alutov, 2021-01-13:
Шум замечал при кипячении или наборе температуры в режиме подогрева, а вот при поддержании температуры сяоми как раз не шумит как по мне. Если почитать [первоисточник](https://github.com/aprosvetova/xiaomi-kettle), то там помимо других всяческих ограничений есть еще и быстрое изменение температуры. Так что выключать питание чайника не самая хорошая идея. В момент выключения (снятии с подставки) он наверняка запоминает температуру, и, если в момент включения она будет сильно отличаться, подогрев не включится. Я пробовал оставлять чайник в режиме Idle. Температура дома где-то 21-23 градуса, чайник остывал до 24-25 и потом включался . Возможно, что 20 градусов и ниже уже слишком мало для включения. Точнее скажут только инженеры сяоми. Сам режим Idle в шлюзе изначально задумывался для оперативного дистанционного выключения в том числе из соображений безопасности. Это потом заметил, что можно и включать чайник, но опять же все очень ненадежно и от многого зависит, о чем и написал в описании шлюза.

closed 2021-09-24 by alutov