alutov/ESP32-R4sGate-for-Redmond #27: Проблема с прошивкой

danilov7624, 2021-01-14:
Добрый день!
Имеется модуль Live D1 mini ESP32. При прошивке вроде как никаких проблем не возникает. Однако затем устройство никак себя не проявляет. К роутеру не подключается. При попытке подключиться к порту с помощью монитора порта получаем кучу значков псевдографики (на сколько я понимаю проблема скорее всего в шрифтах, но перебор шрифтов в приложении ничего не дает). Пробовал на двух подобных модулях, абсолютно новых, результат одинаков.
Как определить в чем может быть проблема?

alutov, 2021-01-15:
О прошивке шлюза на esp32 с картинками можно почитать [здесь](http://zen.yandex.ru/media/id/5f5bea45267c75477b342dab/integraciia-chainika-redmond-g212s-v-umnyi-dom-home-assistant-5fbe19174b9b1b331dae7458)

danilov7624, 2021-01-15:
Так я так и делаю. Но не взлетает почему то. Вот можно бы было сразу свою сеть прошить или бы ESPешка сама бы поднимала web страницу, как обычно это и происходит было удобнее.

alutov, 2021-01-15:
Значит что-то делаете не так или шьете не тот файл. Шить нужно fr4sGate.bin с адреса 0x0.

danilov7624, 2021-01-15:
Ну вот 1 файл это процесс прошивки а 2 файл это то что выдает монитор порта. И никто никуда подлючаться не хочет.
[image]
[image]

alutov, 2021-01-15:
Попробуйте прошить в режиме DIO.

danilov7624, 2021-01-15:
Перепробовал уже все режимы, результат один. Единственное, сейчас обратил внимание, что перед тем, как программа начинает загружать прошивку всплывает следующая строка - AttributeError: 'Panel_Combine' object has no attribute 'choice_Crystal'. 
Это на что то влияет?

alutov, 2021-01-15:
В мониторе порта неверные параметры. Нужно выставить скорость 115200, 8 бит, 1 стоп без контроля четности. После прошивки снять перемычку с IO0 и открыть в роутере либо смартфоне сеть r4s с паролем 12345678.

danilov7624, 2021-01-15:
лог теперь выглядит так (с монитора)  
[log.txt](https://github.com/alutov/ESP32-R4sGate-for-Redmond/files/5821454/log.txt)
Но подключения к роутеру все равно нет.

danilov7624, 2021-01-15:
пробовал открывать сеть и на роутере и на телефоне.

danilov7624, 2021-01-15:
вот так крутиться по кругу. это нормально?

alutov, 2021-01-15:
В логе ошибка [Brownout detector was triggered](http://wikihandbk.com/wiki/ESP32:%D0%A2%D0%B8%D0%BF%D0%BE%D0%B2%D1%8B%D0%B5_%D0%BF%D1%80%D0%BE%D0%B1%D0%BB%D0%B5%D0%BC%D1%8B/%D0%9E%D1%88%D0%B8%D0%B1%D0%BA%D0%B0_%C2%ABBrownout_detector_was_triggered%C2%BB). Плохое питание похоже.

danilov7624, 2021-01-15:
ну это возможно, лог же пишется в случае если есть прямое подключение к компу, а в данном случае это ноутбук, и видимо питания ему не хватает. но в случае подключения к внешнему бп я вообще не понимаю как его можно проверить, ведь к сети он все равно категорически не подключается.

alutov, 2021-01-15:
Это аппаратная проблема. Наберите в гугле и почитайте об этой ошибке. Если напряжение падает ниже минимума, а это чаще всего бывает при старте wifi и/или ble, процессор отключается/рестартует. О каком подключении может идти речь? Меняйте блок питания, а до этого не о чем говорить вообще.

danilov7624, 2021-01-18:
Добрый день!
В общем проблема кроется с том, что модули ESP32 одинаковые, но обвязка и разводка на разных платах разная.
На тех модулях, которые я пытался запустить вначале, происходит какой то сбой ( Live D1 mini ESP32). Сегодня специально купил другой модуль (ESP32 Devkit V1), так вот на нем все заработало сразу, с теми же шнурами и блоками питания, с которыми не хотел работать первый модуль.

closed 2021-07-04 by alutov