alutov/ESP32-R4sGate-for-Redmond #28: Поддержка мультиварки RMC-M903-S

miv777, 2021-01-25:
Подскажите. Добавил по вашей инструкции ранее успешно и полнофункцмонально чайник Redmond RK-G210S. Супер, работает как часы, а сполсденими добавлениями прошивки, теперь и отвалы раз в сутки исчезли!
Шикарно.
Обзавелся так же по случая мультиваркой Redmond RMC-M-903S.
При добавлении выбрал, единственную модель M-800S и, добавилось и даже работает, именуется в шлюзе как RFS-KMC005. Есть сущности, которыми можно включать, выбирать время и задавать программы.
[image]
[image]
[image]
Единственная загвоздка, кроется в несоответствии программ, да и время задавать приходиться как температуру в отдельной сущности. Можно ли это как то скорректировать, в соответствии с реальностью ?
Если нужны логи или какая либо обратная связь, готов предоставить ) либо могу сам руками поправить, только подскажите где )

alutov, 2021-01-26:
Мультиварка управляется из веб и по мктт? Видно состояние, температура, время, работает режим подогрев? Если все ок, нужно только поправить программы. Напишите список программ, добавлю эту модель. По температуре пока скорректировать вряд ли получится, нет в Mqtt Discovery ничего более подходящего.

miv777, 2021-01-26:
Управляется и из веб и по mqtt из HA без проблем. Можно выставить время, выбрать программу и стартануть ее в соответствии с вашими инструкциями =)
Состояние пробрасывает актуальное, температуру текущую не показывает, но целевую можно до начала программы указать, принимает, время, пусть и в цельсиях, соответственно, но отображает текущее, актуализирует в соответствии с текущими показаниями на панели, подогрев активирует и он, разумеется работает.
В каком виде нужен список программ )?
С панели мультиварки ? Или соответствие те что есть сейчас в вебе = тем что выбираются на панели мультиварки ?

alutov, 2021-01-26:
Можно просто порядок программ, 1 - то-то и т.д. И в каких программах есть параметры. Например варка - овощи, рыба, мясо.

miv777, 2021-01-28:
0.climate.r4s2_cooker_temp =on/off Мультиповар (0:30) 100с.(можно менять)
1.Rice = молочная каша (0:35)
2. Slow cooking =Тушение (3:00)
3. Pilaf = Жарка (1:00)
4. Frying_vegetables = Суп (0:15)
Frying_fish = Суп (0:15)
Frying_meat = Суп (0:15)
5.Stewing_vegetables = На пару (1:00)
Stewing_vegetables = На пару (1:00)
Stewing_vegetables = На пару (1:00)
6. Pasta=Макароны (0:08)
7. Milk_porridge= Томление (0:35)
8. Soup =Варка (1:00)
9. Yogurt = Выпечка (8:00)
10. Baking = Крупы (0:45)
11. Steam_vegetables = Плов (0:40)
Steam_fish = Плов (0:40)
Steam_meat = Плов (0:40)
12. Hot = Йогурт (0:40)
13? =Пицца (0:25)
14?= Хлеб (3:00)
15?= Десерты (0:20)

В скобках, время выставляемые по-умолчанию, при выборе этой программы как из ХА, так и с панели мультиварки, в параметрах окна ХА, при этом оно при выборе программы не отображается, но при ее старте, корректно видно и идёт на панели устройства.
Мультиповар включается свитчом и после можно послать нужную температуру/время и устройство его примет, можно будет стартануть Мультиповар включив "обогрев".
13,14,15 программы не выбрать, т.к. в списке закончились опции на последней HOT.
Выбор разновидностей 4,5,11, все равно ведёт у меня к одной и той-же программе, по сути они излишни, т.к. у меня выбор продукта для программ отсутвует в модели.
Как и можно ли активировать отложенный старт не нашел.
Как включить программу "Экспресс"

alutov, 2021-02-01:
Спасибо за информацию, буду разбираться.

alutov, 2021-02-11:
Немного был занят, делал счетчик zmai-90. Продолжаем). Делаю все кроме мультиповара и 16 режима, там пока не все ясно. Есть просьба пройтись по режимам еще раз, только с панели, (лучше с выключенным шлюзом, чтобы он не мешал) и еще раз глянуть температуру и длительность каждого режима. Все цифры выше неверные, их подставлял шлюз из списка программ для 800 мультиварки.

miv777, 2021-02-17:
Ок, выключил шлюз и шел чисто через приложение (температуру видно только в нём) собирая настройки по-умолчанию выставленные для каждой программы.
Вот такой список получился:
0.climate.r4s2_cooker_temp =on/off **Мультиповар** (0:30) 100с.(можно менять)
1.Rice = **Молочная каша** (0:10) Температура 97°C
2. Slow cooking =**Тушение** (1:00) Температура 100°C 
3. Pilaf = **Жарка** (0:15) Температура 170°C 
4. Frying_vegetables = **Суп** (1:00) Температура 99°C
5.Stewing_vegetables = **На пару** (0:20) Температура 100°C
6. Pasta= **Макароны** (0:08) Температура 100°C
7. Milk_porridge= **Томление** (5:00) Температура 97°C
8. Soup = **Варка** (1:00) Температура 100°C
9. Yogurt = **Выпечка** (8:00) Температура 145°C
10. Baking = **Крупы** (0:35) Температура 100°C
11. Steam_vegetables = **Плов** (1:00) Температура 110°C
12. Hot = **Йогурт** (8:00) Температура 38°C
13? =**Пицца** (0:25) Температура 150°C
14?= **Хлеб** (3:00) Температура 150°C
15?= Десерты (0:20) Температура 98°C
16? = Экспресс (Выбрать на панели её можно отдельной кнопкой на панели, отдельной иконки не имеет.

Во всех режимах доступна включенная по-умолчанию опция "Подогрев" и выключена опция "Отложенный старт" (мало-ли это пригодиться для информации), кроме режимов Йогурт (там есть только "отложенный старт"), "Жарка" (там есть только включенный по-умолчанию "подогрев") и "Макароны" (там нету доступных опций.

alutov, 2021-02-18:
Спасибо за данные. Единственное, длительность выпечки вроде бы 1 час, а не 8 часов, смотрел в ютубе. Температуры там нет, это да. Можно пробовать.

miv777, 2021-02-18:
> Спасибо за данные. Единственное, длительность выпечки вроде бы 1 час, а не 8 часов, смотрел в ютубе. Температуры там нет, это да. Можно пробовать.
Блин, да, ложанул, сейчас ещё раз пересмотрел, значение забыл поменять.
Вот дефолтный скрин с выпечки =) разумеется.
[image]

alutov, 2021-02-18:
Можно пробовать уже)

miv777, 2021-02-18:
Только что, как раз проверил, ни один режим после йогурта не выбрать  (пицца, хлеб, десерты)
В веб-панели аналогичным образом, просто ничего не происходит при их выборе.
Выставляется ли верная температура загадка, т.к. окно с режимам при выбранном режиме выглядит вот так
[image]
[image]

Но вообще в целом, если это в интерфейсе/автоматизации скомпоновать, то можно выбрать режим, запустить и выключить при необходимости =)
Так что можно сказать работает, осталось задействовать невклюяащиеся режимы.
Если нужна какая-то обратная связь, буду рад предоставить.

alutov, 2021-02-19:
Я так и не понял, режим йогурт работает? По картинке это не видно. Если режим работает, то при его выборе и нажатии кнопки "Ok" должны появиться температура и время работы по умолчанию. Возможно, что последние 3 режима просто идут не по порядку. Проверить это можно. Нужно перебирать все режимы на самой мультиварке и смотреть код каждого режима на главном экране шлюза в json  строке после "mode:"

miv777, 2021-02-19:
Режим йогурт работает и все что по порядку идут до него.
Не включить и не работают режимы "пицца", "хлеб" и "десерты".
Их код в строке mode, как и остальные значения, несмотря на выбранный режим/режим работы на панели самой мультиварки, к сожалению неизменно отображается вот так
[image]
 т.е. я так понимаю проблема в обратном получении значений в шлюз. Он не меняет ни значение mode, ни значение temp, ни set time, все так и остаётся по нулям, хотя сама мультиварка режим выбирает и даже если его запустить, строка в шлюзе не изменна.

alutov, 2021-02-19:
Это если выбрать 903? А если выбрать 800, состояние показывает? Если да, то похоже не последняя версия. Последнюю версию заливал вчера вечером, возможно нужно скачать ее и обновиться еще раз.

alutov, 2021-02-28:
Мультиварка RMC-903S добавлена в проект.

closed 2021-09-24 by alutov