alutov/ESP32-R4sGate-for-Redmond #29: Поддержка духового шкафа RO-5707S

enigmaspb, 2021-02-01:
Подскажите как вам помочь добавить поддержку данной печки в вашем гейте ?
Какие нужны для этого логи и данные ?

alutov, 2021-02-13:
Можно, но только нужен лог команд и ответов. Если коротко, нужно включить в смартфоне в опциях разработчика лог блютуса, соединиться с печкой и пробежаться по командам. Потом слить лог и далее с ним разбираться. Если будет этот лог, то можно будет добавить печку.

enigmaspb, 2021-02-20:
Сделал лог пробежавшись по всем пунктам меню и включая-выключая режимы  . 
Большое вам спасибо за вашу работу .

[btsnoop_hci.log](https://github.com/alutov/ESP32-R4sGate-for-Redmond/files/6014175/btsnoop_hci.log)

[image]
[image]

alutov, 2021-02-20:
Это тот лог? Я не нахожу данных, только репорты сканирования устройств(advertising reports). Хотя печка мелькает. Чтобы не засорять ветку, можно перейти в telegram. (@alutov).

enigmaspb, 2021-05-23:
Извините давно вам не писал 
пока тот функционал что сделали работает , вот пришла новая ESP с антеной ,  сделал даже экран . 
Правда девушка просит больше информации на экране 

[image]

closed 2021-12-10 by alutov