alutov/ESP32-R4sGate-for-Redmond #31: Термометр Xiaomi Mijia 2

Sinicin1803, 2021-02-13:
Здравствуйте уважаемый Андрей.
Спасибо Вам за замечательную прошивку! Теперь Алиса может мне и чайник вскипятить ;-)

Не можете ли Вы сделать возможность подключения температурных датчиков Xiaomi (в частности  Xiaomi Mijia 2) через Ваш шлюз? А то ставить ещё одну железку - ну просто розеток не хватит ;-) Да и два "порта" как бы свободны остаются, у меня только чайник...

Собственно на ESP32 и делается блютус шлюз - https://www.youtube.com/watch?v=J04PZJTwnnw&t=614s

alutov, 2021-02-27:
Пока датчиками не занимался. Как я понимаю, там данные снимаются в режиме сканирования без установления соединения. Пока не придумал как это все совместить.

Sinicin1803, 2021-02-27:
Был бы весьма благодарен, если бы получилось у Вас всё это совместить.

exotsk, 2021-03-09:
> Пока датчиками не занимался. Как я понимаю, там данные снимаются в режиме сканирования без установления соединения. Пока не придумал как это все совместить.

Да так и есть. Многие bt датчики в том числе мифлора тупо отдают данные в эфир. они зашифрованы, и одно из самых простых решений это шлюз сяоми3 с мод прошивкой, которая организует меш из них, и отдает дешифрованые данные на mqtt сервер на шлюзе.

Sinicin1803, 2021-03-10:
Да я читал про отдельный шлюз. Но была надежда, что не придётся на кухню ещё один модуль ставить... И так уже на кухне под потолком 4 устройства в розетке, придётся удлинитель ставить, не говоря что покупать саму плату (и дело не в цене, долго идёт с Али)...

hc2hunter, 2021-05-17:
> Собственно на ESP32 и делается блютус шлюз - https://www.youtube.com/watch?v=J04PZJTwnnw&t=614s

А чем не устраивает шлюз на прошивке WiFi-IoT ?
Лично у меня уже больше года эти термометры работают именно через него. Никакой Хiaomi-шлюз не нужен.
https://wifi-iot.com/p/wiki/227/

Sinicin1803, 2021-05-17:
Да вся в проблема в количестве железа, которое надо под всё это... На каждый чих - по железке, не многовато ли будет? Вот  и надеялся, что хоть третью не придётся ложить.

alutov, 2022-04-29:
Добавил термометр, но только для прошивки от pvvx в режиме custom.

closed 2022-05-01 by alutov