alutov/ESP32-R4sGate-for-Redmond #33: RK-M216S-E kettle

mbatorowicz, 2021-02-24:
Привет,
есть шанс, что ворота поддерживают RK-M216S-E.
Пожалуйста.

alutov, 2021-02-25:
Да.

mbatorowicz, 2021-02-25:
На данный момент я не вижу эту модель в списке доступных.
Будет ли обновление ПО?
Надо ли самому что-то переделывать?

alutov, 2021-02-27:
Как правило, протокол всех современных моделей чайников Redmond совпадает. Можно выбрать в шлюзе 200 или 240 чайник. Это один и тот же протокол, просто взята нижняя и верхняя модель. Всех поддерживаемых моделей в списке никогда не будет - это никому не нужное расходование ресурсов esp32.

mbatorowicz, 2021-02-27:
Не знаю, как подключить чайник. У меня есть чайник RK-M216S-E. В настройках выбрал RK-G200S. Чайник не рвется. Какая процедура подключения?

alutov, 2021-02-28:
Почитайте для начала главную страничку, там все написано.

mbatorowicz, 2021-02-28:
Мне кажется, что я следую инструкции. Я снял фильм.
https://youtu.be/INdXLsM7BVY

alutov, 2021-02-28:
Шлюз ищет чайник по имени. Вбили произвольное имя. У вас это "czajnik". Это поле должно быть не пустое, иначе скан не запустится. Выбираете сразу 200 или 240 тип. Дальше перегружаете шлюз. Он показывает имя чайника в строке "BLE last found device name".  Копируем(Ctrl-C)  это имя затем идем в Setting и вставляем это имя (Ctrl-V) в поле "Name". Туда, где на видео вписан "czajnik". Сохраняете еще раз. Шлюз перегрузится и только после этого зажимаете и держите кнопку на чайнике.)

mbatorowicz, 2021-02-28:
Спасибо большое - сработало

mbatorowicz, 2021-02-28:
Есть еще одна проблема. Шлюз не может подключиться к серверу MQTT. Все данные обязательно введены правильно: адрес, порт, логин, пароль. У меня есть брокер Mosquitto в Home Assistant. Остальные устройства подключаются без проблем.

alutov, 2021-02-28:
Может в пароле ошибка и хром что-то свое поставил. Остальное видно в установках. Либо шлюз и брокер в разных сетях и они не видят друг друга. Можно с компьютера, где крутится брокер дать пинг на шлюз. Может какое шифрование, я там сильно не вникал, но пока это первый случай проблем с мктт брокером.  Я для тестов, в том числе мктт дискавери, держу на 3 малинке ХА с москитто. Ставил все по умолчанию, и шлюз нормально дружит с москитто.

mbatorowicz, 2021-03-08:
У меня все время проблема с MQTT. Я сделал новый чистый домашний помощник и Mosquito, но он все равно не подключается, когда установлен пароль. Когда я разрешаю подключения без пароля, он подключится. У меня есть # в моем пароле, если это может повлиять на связывание.
Другая проблема заключается в том, что шлюз не отправляет ничего после mqtt. Я использую MQTT Explorer, и он ничего не видит. MQTT.fx тоже ничего не видит.

alutov, 2021-03-09:
Проблема именно в символе "#" в пароле. Вот лог с паролем "esp#":
I (4189) R4S: Mqtt url: mqtt://esp:esp#@192.168.1.10:1883
E (4199) MQTT_CLIENT: Error parse uri = mqtt://esp:esp#@192.168.1.10:1883
W (4199) MQTT_CLIENT: Client asked to stop, but was not started
E (4209) MQTT_CLIENT: Client was not initialized
Помочь не смогу, так как этот символ в  url не принимает сама библиотека Mqtt. Проще его убрать из пароля.

mbatorowicz, 2021-03-09:
Наконец-то это сработало. Спасибо за вашу помощь и терпение.

m4rlb0r0, 2021-03-19:
сотни попыток, и я не могу подключиться ... пожалуйста, дайте мне подсказку!

[image]

alutov, 2021-03-20:
А что стоит в Setting?  Должно быть так:  Name RK-M216S-E (эту строчку лучше скопировать из "BLE last found device name"), тип RK-G200S или RK-G240S. И потом зажать "+" на чайнике до соединения.

m4rlb0r0, 2021-03-20:
[image]

m4rlb0r0, 2021-03-20:
[image]

Скопировал, добавил и зажал кнопку и ничего. Я собираюсь записать видео сегодня вечером

m4rlb0r0, 2021-03-20:
https://youtu.be/ORNd3WLphas

m4rlb0r0, 2021-03-20:
как долго нужно удерживать эту кнопку?

alutov, 2021-03-21:
Либо до соединения, либо как пискнет сразу отпустить. Помнится, на чайнике у меня проходил первый вариант, на розетке только второй. У розетки, правда, звука нет, как начнет мигать нужно отпускать кнопку.

m4rlb0r0, 2021-03-23:
для сопряжения этой модели нужно удерживать кнопку включения нет +

оно работает ! большое спасибо !

closed 2021-09-24 by alutov