alutov/ESP32-R4sGate-for-Redmond #34: Почему не заходит на сайт?

fr00kt, 2021-02-28:
Почему не заходит на сайт?
Вот логи ESP32

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7024
ho 0 tail 12 room 4
load:0x40078000,len:14440
ho 0 tail 12 room 4
load:0x40080400,len:3980
entry 0x40080680
I (32) boot: ESP-IDF v4.2-dev-1905-g625bd5eb1 2nd stage bootloader
I (32) boot: compile time 11:26:29
I (32) boot: chip revision: 1
I (37) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (44) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : DIO
I (53) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (81) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (89) boot:  2 otadata          OTA data         01 00 0000e000 00002000
I (96) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (104) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (111) boot:  5 spiffs           Unknown data     01 82 003d0000 00030000
I (119) boot: End of partition table
I (123) boot_comm: chip revision: 1, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x54638 (345656) map
I (270) esp_image: segment 1: paddr=0x00064660 vaddr=0x3ffbdb60 size=0x05850 ( 22608) load
I (279) esp_image: segment 2: paddr=0x00069eb8 vaddr=0x40080000 size=0x00400 (  1024) load
I (280) esp_image: segment 3: paddr=0x0006a2c0 vaddr=0x40080400 size=0x05d58 ( 23896) load
I (298) esp_image: segment 4: paddr=0x00070020 vaddr=0x400d0020 size=0x10ecf8 (1109240) map
I (717) esp_image: segment 5: paddr=0x0017ed20 vaddr=0x40086158 size=0x18818 (100376) load
I (777) boot: Loaded app from partition at offset 0x10000
I (777) boot: Disabling RNG early entropy source...
I (778) cpu_start: Pro cpu up.
I (781) cpu_start: Application information:
I (786) cpu_start: Project name:     r4sGate
I (791) cpu_start: App version:      1.0
I (796) cpu_start: Compile time:     Jan 11 2021 16:53:18
I (802) cpu_start: ELF file SHA256:  a57f614a60c784d4...
I (808) cpu_start: ESP-IDF:          v4.2-dev-1905-g625bd5eb1
I (814) cpu_start: Starting app cpu, entry point is 0x40081360
I (0) cpu_start: App cpu up.
I (825) heap_init: Initializing. RAM available for dynamic allocation:
I (831) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (837) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (844) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (850) heap_init: At 3FFD9468 len 00006B98 (26 KiB): DRAM
I (856) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (862) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (869) heap_init: At 4009E970 len 00001690 (5 KiB): IRAM
I (875) cpu_start: Pro cpu start user code
I (893) spi_flash: detected chip: generic
I (894) spi_flash: flash io: dio
I (894) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Starting r4sGate...
This is esp32 chip with 2 CPU cores, WiFi/BT/BLE,
silicon revision 1, 4MB external flash
I (922) R4S: Init start free heap: 158076

APP IDF version: v4.2-dev-1905-g625bd5eb1
I (2172) R4S: ILI9341 TFT not found.
I (2172) BTDM_INIT: BT controller compile version [826d7c6]
I (2172) system_api: Base MAC address is not set
I (2172) system_api: read default base MAC address from EFUSE
I (2272) phy: phy_version: 4182, f1ba940, Jun  4 2020, 19:40:07, 0, 0
I (2512) R4S: app_main init bluetooth

I (2682) wifi:wifi driver task: 3ffe308c, prio:23, stack:6656, core=0
I (2702) wifi:wifi firmware version: 0507408
I (2702) wifi:wifi certification version: v7.0
I (2702) wifi:config NVS flash: enabled
I (2702) wifi:config nano formating: disabled
I (2702) wifi:Init dynamic tx buffer num: 32
I (2712) wifi:Init data frame dynamic rx buffer num: 32
I (2712) wifi:Init management frame dynamic rx buffer num: 32
I (2722) wifi:Init management short buffer num: 32
I (2722) wifi:Init static rx buffer size: 1600
I (2732) wifi:Init static rx buffer num: 10
I (2732) wifi:Init dynamic rx buffer num: 32
I (2752) wifi:mode : sta (b4:e6:2d:b3:f0:25)
I (2752) R4S: Wifi_init_sta finished.
I (2752) R4S: Connecting to Wifi: 'r4s' with password '12345678'
I (2872) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (2882) wifi:state: init -> auth (b0)
I (2892) wifi:state: auth -> assoc (0)
I (2892) wifi:state: assoc -> run (10)
I (3322) wifi:connected with r4s, aid = 8, channel 1, 40U, bssid = 2a:6c:07:44:65:d2
I (3332) wifi:security type: 2, phy: bgn, rssi: -33
I (3332) wifi:pm start, type: 1

I (3422) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4162) esp_netif_handlers: sta ip: 192.168.31.89, mask: 255.255.255.0, gw: 192.168.31.1
I (4162) R4S: got ip:192.168.31.89
I (4162) R4S: Connected to ap SSID:'r4s' password:'12345678'
E (4172) MQTT_CLIENT: Error parse uri = mqtt://:@:1883
W (4172) MQTT_CLIENT: Client asked to stop, but was not started
E (4182) MQTT_CLIENT: Client was not initialized
I (4192) R4S: Init end free heap: 61684
```

alutov, 2021-02-28:
Видно что к wifi сети присоединился. Пинг проходит на 192.168.31.89? Есть вариант, что гостевая в роутере сеть полностью изолирована от домашней.

fr00kt, 2021-02-28:
Пинг не проходит, пробовал на телефоне точку доступа поднять тоже на сайт не заходит

fr00kt, 2021-03-02:
Я записывал такой командой `esptool.py --port COM3 write_flash --flash_mode dio --flash_size 4MB 0x0000 fr4sGate.bin --erase-all` Правильно?

alutov, 2021-03-05:
Если к сети присоединился, то прошит вроде бы правильно. Но почему не открывается страница в браузере и пинг не проходит сказать тяжело.

closed 2021-09-24 by alutov