alutov/ESP32-R4sGate-for-Redmond #35: Перезагрузка шлюза

DivanX10, 2021-03-05:
Вы написали, что можно перегрузить шлюз и по Mqtt, записав в топик screen restart, reboot или reset. Не могли бы уточнить как это делать? Я правильно понимаю, что нужно указать r4s/devaddr/cmd/restart? Можете это добавить в топик, чтобы в Home Assistant можно было перезагрузить шлюз?

alutov, 2021-03-05:
В топик "r4s/screen" нужно ввести "restart", "reboot" или "reset". Так как рестарт часто не нужен, отдельный топик не заводил.

DivanX10, 2021-03-06:
Сделал такой скрипт в Home Assistant для перезапуска шлюза и все работает. Спасибо. 

```
alias: Перезапуск шлюза esp r4s
sequence:
  - service: mqtt.publish
    data:
      topic: r4s/screen
      payload_template: restart
```

closed 2021-03-06 by DivanX10