alutov/ESP32-R4sGate-for-Redmond #36: Поддержка устройства: увлажнитель воздуха RHF-3320S

ChaoticBD, 2021-03-05:
Доброго времени суток!

прошу добавить устройство:  https://store.redmond.company/ru/m/catalog/smart_tech/umnyy-uvlazhnitel-vozdukha-redmond-skydew-3320s/

мануал https://mcgrp.ru/files/viewer/685768/7#navigate_bar

Думал помочь и добавить, но после java код просто нечитаемый, так что сорри =с 

Протокол:
 55
 1d
 06
 00		авто режим  - 01 вкл / 00 выкл  
 00		интенсивность увлажнения: 00 - 1, 01 - 2, 02 - 3, 03 - 4
 28 	поддерживаемая влажность в десятеричной системе 40/45/50/55/60/65/70/75/80 (при включенном авто режиме)
 00 	--
 00 	--
 00 	--
 00 	таймер выключения часы в десятеричной системе
 00 	таймер выключения минуты в десятеричной системе
 01		устройство - 01 вкл / 00 выкл 
 1a     влажность в десятеричной системе
 00 	--
 1a     температура в десятеричной системе
 00 	--

 00		ночной режим - 01 вкл/00 выкл
 01 	--
 00		04 - нет воды/поднята крышка
 aa 	--

значения с -- не получилось как то изменить.
В рамках предпоследнего "00" -подозреваю там еще могут быть какие либо ошибки, попробую вызвать их.		
Если будет что дополнить - отпишусь

alutov, 2021-03-05:
Ага, почитаю, но займусь уже после праздников.

ChaoticBD, 2021-07-02:
День добрый!
Предполагается какая нибудь доработка? Или можно не ждать? х)

alutov, 2021-07-02:
Не делал, так как не понял протокол. Я так понимаю, тут только ответ на запрос состояния. А сами команды установки всех режимов какие? Проще всего это сделать в родном приложении на андроиде, предварительно включив лог BT, перебирая все режимы. По логу уже можно что-то делать.

ChaoticBD, 2021-07-09:
ок, сейчас дома ремонт и устройство вне доступа. как будет возможность - запишу

closed 2023-10-14 by alutov