alutov/ESP32-R4sGate-for-Redmond #37: Отвал BT

exotsk, 2021-03-12:
Андрей, поставил от 8.03. с  IDF v4.4-dev-4-g73db14240, стабильный отвал BLE раз в сутки. Одновременно улучшил качество wifi. не помогло.

alutov, 2021-03-12:
У меня пока все нормально. Wifi правил в 9.03.2020. Если отвалы остались, придется поставить птичку в  "BLE monitoring".

exotsk, 2021-03-12:
Понял тебя. Я сегодня 9.03 поставил. Пока сутки не прошли. Отпишусь.

alutov, 2021-03-12:
А сколько устройств определено и подключено? У меня 3 и 2 на одном, 2 и 2 на другом. То есть скан постоянно не бегает, все соединилось и скан остановился. Может в этом дело.

exotsk, 2021-03-16:
1 устройство. 201 чайник, есть еще мультиварка но она кардинально не совпадает с твоей :) 
Только что залипла предыдущая версия с галкой BLE monitoring
сейчас поставлю  2021.03.14. 
скажи как лог снять, сделаю. Если уарт слушать, то на таком удалении у меня не получится. чайник далеко от компа.

alutov, 2021-03-16:
Пока ничего не нужео. Нашли один баг в новой в4рсии. Может поможет. И добавил строку состояния скана, видна до соединения.

exotsk, 2021-03-20:
По кинетикам, второй шлюз стоить в паре метров по за стеной. Знаю о том что можно указывать точку в меше, установил принудительно ближнего. Мультиварка SkyCooker M223S

exotsk, 2021-04-06:
Андрей, последняя версия похуже будет в стабильности, 2 суток подряд отвалы.
[image]

alutov, 2021-04-07:
Версия от 25.03? В чем проблема, зависание бле, перезагрузка? Что показывает ble activity, scanning, idle? Пока жалоб на отвал бле не было. У меня 9 суток аптайм без отвалов, причем ble monitoring выключен. Код в последних версиях я не менял, только опции сборки.

exotsk, 2021-04-07:
значит так совпало. Но галка стоит, чайник на базе, чайника не видит. Самая последняя, которая выложена сейчас. сейчас сутки вроде норм

closed 2021-07-04 by alutov