alutov/ESP32-R4sGate-for-Redmond #38: Температура 35

dmatora, 2021-03-12:
Купил чайник Redmond (в основном чтобы греть воду до 35 градусов)
В приложении ReadyForSky такая возможность есть, а вот в R4sGate все что меньше 40 - отключает подогрев
Нельзя ли добавить поддержку низких температур?

alutov, 2021-03-13:
Какая минимальная температура подогрева в этом чайнике?

dmatora, 2021-03-13:
35
[image]

alutov, 2021-03-13:
Сделал минимум 30 по mqtt, только для редмонда. Xiaomi забросили? Я только на днях обнаружил, что в нем keep warm перезапускается только один раз...

dmatora, 2021-03-13:
Xiaomi так и не удалось решить проблему с потерей контроля при остывании воды. У меня 2 xiaomi чайника, под разные задачи, без удаленного управления сильно не страдал. 
Redmond купил только когда стала нужна вода 35 градусов, плюс долго мониторил когда на Авито хорошая цена подвернется (взял новый за 2000р)

alutov, 2021-03-13:
Наверное там таки 20 градусов предел. Все же хочу добить хотя бы бессрочный keep warm.

dmatora, 2021-03-13:
У меня ниже 30 (если не 40) контроль теряется вроде

dmatora, 2021-03-13:
В помещении у меня не ниже 20

dmatora, 2021-03-18:
> Сделал минимум 30 по mqtt, только для редмонда. Xiaomi забросили? Я только на днях обнаружил, что в нем keep warm перезапускается только один раз...

Попытался обновить прошивку через веб, не получилось.
Накатил через flasher, 35 градусов работает, спасибо.

closed 2021-03-18 by dmatora