alutov/ESP32-R4sGate-for-Redmond #39: Стоит явно упомянуть ограничение на длину пароля MQTT

strannick-ru, 2021-03-16:
При попытке ввести пароль на 16 символов (у меня как раз такой использовался), он не сохраняется после перезагрузки и соединения с MQTT-брокером не происходит.

Если уменьшить длину пароля  всё ок.

Выяснил, что причина именно в этом не сразу  : )
Думаю, стоит задокументировать это.

alutov, 2021-03-16:
Есть такое, длина 15 символов, вайфая 63. Не лумал, что в локалке сильно будут шифроваться)))

alexreddy78, 2021-08-19:
Добрый день!

Никак не поправить? Уж очень не хочется в MQTT пароль менять и везде потом его переделывать.

Спасибо

alutov, 2021-09-04:
И какая длина пароля нужна?

alexreddy78, 2021-09-05:
> И какая длина пароля нужна?

Добрый день, Андрей!

Ну хотелось бы 18 символов :)

С Уважением

alutov, 2021-09-22:
Сделал 19 символов.

alexreddy78, 2021-09-23:
@alutov Благодарю! Думаю автор issue должен закрыть его :) @strannick-ru

closed 2021-09-23 by strannick-ru