alutov/ESP32-R4sGate-for-Redmond #40: Xiaomi Mi smart Kettle PRO

hc2hunter, 2021-03-19:
Возможна ли поддержка этого чайника?
Подробнее проблему изложил тут: https://mjdm.ru/forum/viewtopic.php?f=8&t=5501&start=250#p127307
Посоветуйте, пожалуйста. Спасибо.

alutov, 2021-03-20:
Возможна, но только нужно знать его продукт айди, чтобы внести его в программу. Я не знаю, у него он свой или нет. Попробуйте все 3 варианта Mikettle, может с каким-то и запустится.

hc2hunter, 2021-03-20:
Все 3 пробовал - эффект одинаковый.
Также пробовал ставить Рэдмонд 240 - пищит так же, только реже.
Из Mi HOME от Vevsвытящил следующую информацию по чайнику:
`model   : yunmi.kettle.v12`
`f/w     : 4.1.0_0003`
`did     : blt.6.1633n3fn0sg00`
`uid     : 1775019949`
token, bindkey  и mac вам вряд ли нужны...
Скажите, что где еще что посмотреть - предоставлю информацию.
Спасибо!

alutov, 2021-03-21:
MJHWSH02YM - это написано на дне чайника вместе с Model? Интересное название, сильно отличается от моего. Чтобы найти product id придется запастись терпением. Нужно скачать новую версию r4sGate.bin обновиться в браузере, выбрать  тип "Mi-Unknown" и ждать. Ждать придется долго. Шлюз начнет перебирать product id от 1 до 10000. Видно это будет в строке "BLE activity". У меня перебор от 1 до 275 занял 10 минут. Если чайник был привязан, то при нахождении правильного айди чайник соединится. Если не авторизован, то при достижении правильного айди  он перебор остановится и шлюз постоянно должен пытаться подключиться. Если дойдет до 10000 то похоже метод не сработал. Если метод сработает, потом внесу модель и айди в программу, чтобы не перебмрать каждый раз при рестарте. Не знаю, оставлять ли потом эту фичу.

hc2hunter, 2021-03-21:
да, на дне чайника.

hc2hunter, 2021-03-21:
Как обозвать девайс в настройках не важно? А то сейчас там у меня прописано MiKetv12.
Или стереть имя?

hc2hunter, 2021-03-21:
А... видимо нужно обозвать. Без имени он не сканирует.

alutov, 2021-03-21:
Нужно правильно указать имя, иначе не сработает. Имя скопируйте из ble last found device name. Чтобы проверить правильность айди, нужно именно соединиться с чайником. Для чистоты эксперимента, соединение устанавливается отдельно для каждого айди, потом рвется.

hc2hunter, 2021-03-21:
Ну... пока прошло несколько минут, но я не вижу, чтобы он перебирал ID.
Постоянно вижу на главной странице это:

BLE last found device name | MiKetv12
BLE activity | Connecting MiKetv12 (Open/Id:1)

alutov, 2021-03-21:
Тип чайника проставлен Mi-Unknown?

hc2hunter, 2021-03-21:
[image]
[image]

alutov, 2021-03-21:
ну тогда может warm подержать.

hc2hunter, 2021-03-21:
ну да, стоит на 1.
warm держать сколько? а то он же греть начинает при удержании

alutov, 2021-03-21:
Пискнуть должен. Ну как к михому привязывали?

hc2hunter, 2021-03-21:
Никак. Он сам нашелся и прицепился. Удерживать ничего не нужно.

hc2hunter, 2021-03-21:
Ну как и термометры BLT, например. Их просто видно в сети и все. "Пэйринг" не нужен.

alutov, 2021-03-21:
Странно, у меня нажимать нужно и он привязывается. Очень может быть, что тут другой протокол.

hc2hunter, 2021-03-21:
Попытался удерживать, ничего не происходит.
Если удерживать больше секунды - включается BOIL. Если держать сколько угодно долго - вообще ничего не происходит при отпускании. При этом чайник так и пищит раз в секунду.

hc2hunter, 2021-03-21:
Где смотреть log в интерфейсе не нашел.
 
В отладке каждую секунду происходит так:
[0;32mI (20264) R4S: Searched A device MiKetv12

[0;32mI (20274) R4S: Scan stop successfully
[0;32mI (20274) R4S: Connect A to the remote device
[0;32mI (20384) R4S: Open A success, conn_id 0, if 1, status 0, mtu 23
[0;32mI (20384) R4S: REMOTE BDA A:
[0;32mI (20394) R4S: a4 c1 38 82 49 5a
[0;32mI (22134) R4S: Discover service A complete conn_id 0
[0;32mI (22144) R4S: SEARCH RES A: conn_id = 0 is primary service 1
[0;32mI (22144) R4S: start handle 63 end handle 75 current handle value 63
[0;32mI (22154) R4S: Xiaomi Service1 A found
[0;32mI (22154) R4S: SEARCH RES A: conn_id = 0 is primary service 1
[0;32mI (22164) R4S: start handle 34 end handle 53 current handle value 34
[0;32mI (22164) R4S: Xiaomi Service2 A found
[0;32mI (22174) R4S: SEARCH RES A: conn_id = 0 is primary service 1
[0;32mI (22174) R4S: start handle 12 end handle 16 current handle value 12
[0;32mI (22184) R4S: Xiaomi Service3 A found
[0;31mE (22194) R4S: Get_auth_by_uuid A error
[0;31mE (22194) R4S: Get_ver_by_uuid A error
[0;31mE (22194) R4S: Get_setup_by_uuid A error
[0;31mE (22204) R4S: Get_time_by_uuid A error
[0;31mE (22204) R4S: Get_boil_by_uuid A error
[0;31mE (22214) R4S: Get_mcuver_by_uuid A error
[0;32mI (22224) R4S: ESP_GATTC_DISCONNECT_EVT, reason = 22
[0;32mI (22224) R4S: Scan starting
[0;32mI (22234) R4S: Scan start success
[0;32mI (22344) R4S: a4 c1 38 82 49 5a
[0;32mI (22344) R4S: searched Device Name Len 0

alutov, 2021-03-21:
Можно глянуть лог шлюза, но там только можно проверить, где затык. Если ковырять дальше, нужно в смарте открывать лог блютус, входить в михом и пытаться соединиться. И как-то странно, что цепляется без привязки. Это он к любому левому смарту прицепиться может? Если протокол свой, то еще потом ковырять михом.

hc2hunter, 2021-03-21:
Думаю, что к любому, кто запросит коннект с чайником.
Если это не проделки Mi HOME от VEVS

alutov, 2021-03-21:
В общем свои сервисы в нем.

hc2hunter, 2021-03-21:
в MiHOME когда мешь на чайник, он пищит один раз и появляется картинка.
[image]

в лог vevs он вообще ничего не отписал....
Отметился лишь так:

{
  "did": "blt.6.1633mvt64sg00",
  "uid": 1775019949,
  "token": "62b01594865831630aced7e4",
  "name": "Mi Smart Kettle Pro",
  "pid": 6,
  "mac": "A4:C1:38:82:49:5A",
  "bssid": "",
  "longitude": "0.00000000",
  "latitude": "0.00000000",
  "show_mode": 1,
  "model": "yunmi.kettle.v12",
  "permitLevel": 16,
  "isOnline": false,
  "extra": {
    "isSetPincode": 0,
    "pincodeType": 0,
    "isSubGroup": false,
    "showGroupMember": false
  },
  "orderTime": 1616088993,
  "freqFlag": true,
  "hide_mode": 0,
  "comFlag": 1,
  "api_server": "ru"
}

hc2hunter, 2021-03-21:
Пойду обновлю MiHome... а то стоит 6.0, а уже оказывается есть 6.2

hc2hunter, 2021-03-21:
Никакой новой инфы не появилось. Чайник при общении с ним в логе vevs ничего про себя не оставляет.

alutov, 2021-03-21:
Получается, что теперь только снимать лог по блютусу в смарте с работающей программой. Ну и потом ковырять библикотеки михома. Вторым сам пока не занимался(

alutov, 2021-03-21:
Вот мой лог для сравнения:
[0;32mI (143654) R4S: b8 7c 6f 84 24 04 [0m
[0;32mI (143654) R4S: searched Device Name Len 8[0m
[0;32mI (143654) R4S: MiKettle[0m
[0;32mI (143654) R4S: 
[0m
[0;32mI (143654) R4S: Searched A device MiKettle
[0m
[0;32mI (143664) R4S: Scan stop successfully[0m
[0;32mI (143664) R4S: Connect A to the remote device[0m
[0;32mI (144114) R4S: Open A success, conn_id 0, if 1, status 0, mtu 23[0m
[0;32mI (144114) R4S: REMOTE BDA A:[0m
[0;32mI (144114) R4S: b8 7c 6f 84 24 04 [0m
[0;32mI (144204) R4S: update connection params status = 0, min_int = 6, max_int = 18,conn_int = 18,latency = 0, timeout = 50[0m
[0;32mI (145834) R4S: Discover service A complete conn_id 0[0m
[0;32mI (145834) R4S: SEARCH RES A: conn_id = 0 is primary service 1[0m
[0;32mI (145834) R4S: start handle 56 end handle 73 current handle value 56[0m
[0;32mI (145844) R4S: Xiaomi Service1 A found[0m
[0;32mI (145844) R4S: SEARCH RES A: conn_id = 0 is primary service 1[0m
[0;32mI (145854) R4S: start handle 35 end handle 48 current handle value 35[0m
[0;32mI (145854) R4S: Xiaomi Service2 A found[0m
[0;32mI (145864) R4S: SEARCH RES A: conn_id = 0 is primary service 1[0m
[0;32mI (145874) R4S: start handle 16 end handle 34 current handle value 16[0m
[0;32mI (145874) R4S: Xiaomi Service3 A found[0m
[0;32mI (145884) R4S: Register_for_notify A[0m
[0;32mI (145984) R4S: Write_auth A Ok[0m
[0;32mI (146074) R4S: Write_auth_mi A Ok[0m
[0;32mI (146164) R4S: Write_auth_xi_ack A Ok[0m
[0;32mI (146164) R4S: Write_setup_xi A Ok[0m
[0;32mI (146164) R4S: Write_boil_xi A Ok[0m
[0;32mI (146174) R4S: Write_time_xi A Ok[0m
[0;32mI (146174) R4S: Authorize A Xiaomi ok[0m
[0;32mI (150494) R4S: Notify Data A:[0m
[0;32mI (150494) R4S: 03 02 01 00 28 2e 01 33 00 00 18 01 [0m

closed 2022-02-03 by alutov