alutov/ESP32-R4sGate-for-Redmond #42: Самопроизвольное переключение чайника на кипячение

slydiman, 2021-04-12:
Чайник RK-M171S, в настройках RK-M173S.
Home Assistant с MQTT брокером. 
Раньше (пол года назад) включаешь его (кнопкой на чайнике) на кипячение, а он тут же выключается. Приходится включать повторно.
На днях обновился на 12.01.2021.
Теперь другая проблема - включаешь чайник на подогрев скажем на 80 градусов (кнопками на чайнике), а он переключается на 100, то есть на кипячение.
Никаких retain нигде не стоит. 
В Home Assistant используется discovery (раньше была ручная настройка). 

Кстати, объект climate жутко неудобен из-за того что постоянно сбрасывается температура в 0. 
Чтобы включить подогрев надо каждый раз с нуля крутить до нужного значения. Зачем эти пляски с температурой? 
В идеале нужно всего 2 вещи:
 - выставить температуру (и чтобы она никуда не сбрасывалась), при этом 100 градусов - это 100 градусов, то есть кипячение.
 - вкл/выкл

Я понимаю что в протоколе для кипячения температура может и ноль, но конечным пользователям она зачем? 
К примеру температура стоит всегда на 80. Нажал вкл - подогрел. 
Надо скипятить - поднял температуру до максимума и всё.
В теории в Home Assistant можно на templates попробовать сделать свой правильный объект climate.

Но на сегодня самое главное - решить проблему с переключением на 100. 
Фактически подогрев не работает вообще, хоть откатывайся на прошивку полугодичной давности.

Могу помочь с отладкой и пр.

alutov, 2021-04-12:
На сегодня актуальная версия 2021.03.25, лежит в папке build. С января прошло уже слишком много времени. Пробовал установить подогрев 80 и 90 кнопками на чайнике RK-M173S, все корректно отрабатывает, ничего подобного переключению на кипячение не наблюдается. Изначально шлюз делал исключительно для себя и не под ассистент, а иоброкер. Именно потому используется Mqtt. И уже потом [был добавлен](https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/2) Hass Discovery из того, что предоставлял ассистент по Mqtt. Сущности нагревателя не было, нет ее и сейчас, насколько я знаю, пришлось делать из климата. Уж как получилось. Не нравится - ничего не мешает выключить Hass Discovery и сделать все ручками, так многие делают. Тут я не помощник, с ассистентом мало знаком. Добавят сущность чайника в ассистент, переделаю.

slydiman, 2021-04-12:
Ок, пробую 2021.03.25
Водонагреватель в HA есть, но совсем ущербный https://www.home-assistant.io/integrations/water_heater/
Использование climate - нормально, под него много UI и пр., но надо допилить отдельный режим для кипячения.
Попробую набросать, если нормально получится, то пришлю.

alutov, 2021-04-13:
В Mqtt discovery его как раз и нет. Все, что можно сделать, посылая команды по Mqtt, описано [тут](https://www.home-assistant.io/docs/mqtt/discovery/).

slydiman, 2021-04-13:
Этот вопрос видимо можно закрыть. 
С последней прошивкой пока самопроизвольного переключения не замечено.
Остался другой важный вопрос https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/43

slydiman, 2021-04-17:
Уже после закрытия issue снова наблюдал эту же проблему на последней прошивке.
Кнопками на чайнике выставил 85, нажал подогрев, через пару секунд релюшка сама выключилась и тут же включилась уже на 100 (кипячение). Причем ещё через пару секунд релюшка еще пару раз щелкнула, режим остался кипячение.
Пришлось выключать руками и заново ставить на подогрев. Со второй попытки остался подогрев и само уже никуда не переключалось. У меня лично проблема проявилась всего 1 раз, у жены это срабатывает через раз.
Чайник в 1.5м от шлюза, связь стабильная.

slydiman, 2021-04-27:
Есть подозрение что это связано с логикой `90 и более = кипячение`. Надо сделать ну хотя бы 95, а по хорошему кипячение - это строго 100 без всякой творческой логики. Нужно правильно заварить чай и нужна температура 90 - и это невозможно в принципе пока шлюз не обесточишь!

alutov, 2021-04-27:
Поднял температуру включения кипения до 99.

alutov, 2021-07-04:
Судя по всему проблема ушла.

closed 2021-07-04 by alutov