alutov/ESP32-R4sGate-for-Redmond #43: Режимы работы и MQTT

slydiman, 2021-04-12:
Пытаюсь сконфигурировать нормальный объект climate в Home Assistant.
Можно использовать templates, но интеграция всё равно [climate.mqtt](https://www.home-assistant.io/integrations/climate.mqtt/) как не крути.
Пытаюсь к режимам `off` / `heat` добавить режим `auto` или задействовать опцию `aux heat`.

Сразу скажу что **оптимальный вариант - использовать `aux heat`** в climate для кипячения вместо отдельного переключателя switch.r4s1.Kettle.boil, но рассмотрим и другие варианты.

Столкнулся с проблемами в дизайне шлюза:

Проблема #1: Попытался добавить режим `auto`. 
`mode_command_topic` один на все режимы и не смотря на наличие `mode_command_template` не получается выставить что нужно через единственный топик. Допустим используем раздельные cmd/rsp.
cmd/state <-- 0/off/false - выключение, 1/on/true - кипячение, 2...100 - кипячение и подогрев;  А где просто подогрев???
cmd/heat_temp <-- 0...39 - выключение, 40...90 подогрев, > 90 - кипячение;
Итого просто невозможно изменив значение одного топика выбрать режимы off / heat / boil / boil + heat. 

Проблема #2:  Попытался прикрутить `aux heat` к топику state аналогично как сейчас сделан switch.r4s1.Kettle.boil, но в случае если heat_temp = 85, это ни разу не boil, а просто нагрев. То есть `aux heat` как и switch boil переключается параллельно с режимом off/heat, в чем нет никакого смысла.

Проблема #3: всегда сбивается заданная температура. Просто невозможно держать объект climate в состоянии off и при этом хранить температуру 85.

Предлагаю рассмотреть следующий дизайн:
`heat_temp` - выставляется либо на чайнике кнопками, либо через MQTT и никогда не сбрасывается само вне зависимости ни от каких режимов и состояний. Изменение `heat_temp` через MQTT не должно менять режим (включать/выключать чайник) - это только target temperature, а не переключатель с хитрой логикой.
`hstate` - оставить как сейчас (off / heat) - управляет только подогревом, никакого кипячения.
'state' - `on/off` связать с `aux_command_topic` и `aux_state_topic` в Home Assistant - управляет именно кипячением и имеет приоритет над `hstate`. 
`state` и 'hstate' должны быть независимыми. То есть две boolean переменные `state` и `hstate` дают всё 4 возможных режима работы - выкл / нагрев / кипячение / кипячение + подогрев. К примеру в режиме кипячение + подогрев скипел чайник - `state` (`aux`) выключился, `hstate`=heat остался.
Такой дизайн решает все проблемы и позволит настроить нормальный объект climate, думаю и не только в Home Assistant. В конце концов если в какой-нибудь системе нет aux heat, останется переключатель boil (завязанный на топик `state`), который будет работать правильно, а не так как сейчас.

Пожелание: в Home Assistant discovery в device добавить MAC адрес  "connections": [["mac", "12:34:56:78:90:ab"]]. Это позволит правильно группировать устройства внутри Home Assistant.

slydiman, 2021-04-13:
Пока сделал вот такой объект climate в Home Assistant. 
Костыли конечно, но сейчас этим хотя бы можно пользоваться.
Режимы: off / heat (включается на 85) / auto (кипячение)
```
  - platform: mqtt
    name: Kettle
    retain: false
    min_temp: 0
    max_temp: 100
    temp_step: 5
    precision: 1
    temperature_command_topic: "r4s/xxx/heat_temp"
    temperature_state_topic: "r4s/xxx/heat_temp"
    current_temperature_topic: "r4s/xxx/temp"
    availability_topic: "r4s/xxx/status"
    mode_command_topic: "r4s/xxx/heat_temp"
    mode_command_template: >
      {% if value == "off" %}
      0
      {% elif value == "heat" %}
      85
      {% else %}
      100
      {% endif %}
    mode_state_topic: "r4s/xxx/json"
    mode_state_template: >
      {% if value_json.state == 0 %}
      off
      {% elif value_json.target > 0 %}
      heat
      {% else %}
      auto
      {% endif %}
    modes:
      - "off"
      - "heat"
      - "auto"
```
ну и из всего многообразия UI [вот этот](https://github.com/nervetattoo/simple-thermostat) более менее
```
          - type: custom:simple-thermostat
            entity: climate.kettle
            header:
              name: Чайник
              icon: mdi:kettle
            step_size: 5
            control:
              hvac:
                "off":
                  name: Выкл.
                  icon: mdi:kettle-off
                heat:
                  name: Нагрев # 85°C
                  icon: mdi:kettle-alert
                auto:
                  name: Кипячение
                  icon: mdi:kettle-steam
            label:
              temperature: Сейчас в чайнике
```

RavenHuginn, 2021-05-19:
Что-то не получается в UI добавить карточку, горит красным

alutov, 2021-09-22:
В новой версии изменена логика работы Mqtt топика State в чайнике Redmond. Теперь он управляет только режимом кипячения, не влияя на режим подогрева. Это дает возможность использовать все режимы чайника. Поправлен и топик heat_temp. установка температуры до 30°C выключает подогрев, если он был включен и включает подогрев с последней запомненной в шлюзе температурой. Сделано для того, чтобы меньше ковыряться в интерфейсе ассистента (Вбить температуру напрямую цифрами разработчики ассистента не сочли нужным). С этой же целью установка температуры 100°C выключает подогрев. Что касается via_device. Как написано в [документации](https://www.home-assistant.io/integrations/climate.mqtt/), там нужно указывать идентификатор устройства через которое идет обмен сообщениями с ассистентом. Пробовал использовать его во всех объектах с идентификатором шлюза. Эффекта не заметил. Ведь по логике, если становится недоступен шлюз, должны стать недоступными все зависимые от него объекты. Чего на практике не происходит.

closed 2022-02-02 by alutov