alutov/ESP32-R4sGate-for-Redmond #45: Rs170S

robertbigk, 2021-06-23:
Hi I add Redmond SkyKettle RK-M170S-E

[image]

Kettle detected and work switch on/off

[image]
Kattle  have multiple temperature setup 40 ,55,70,80,95

[image]

when I try setup temperature I set 55 and the function goes off.

only work switch on go heat for 100. How to add to Ha's missing functionality?

Kettle very silimar for 173s  multistep temperature. Coud You help me?

alutov, 2021-06-24:
It may be that in this model the protocol differs from the same 173 teapot. I saw something similar on the majordomo. If the native android r4s application can set only a few temperature values, and not any in the range of 30-90, then it is so.

robertbigk, 2021-06-24:
Thx for the answer ok then possible to create entities or other way option to setup this couple level o temperature?
40,55,70,85 and 95? Or only way I have option heat or no heat and that it?:)

alutov, 2021-06-26:
Upload "r4sGate170.bin" from "build" folder, select rk-m170s and test it in all modes (web and mqtt).

robertbigk, 2021-06-26:
Supert Work!!! Spasiba Balszoje:) 

[image]

First show current temperature befoure that not work .

Second is jumping to step what got kettle i choise 20 his jump to 1 set of temperature 40 i choise 45 jump to 55. I perfect big thanks for help

closed 2021-06-28 by robertbigk