alutov/ESP32-R4sGate-for-Redmond #46: Некорректный JSON

slydiman, 2021-07-21:
Вот тут https://github.com/alutov/ESP32-R4sGate-for-Redmond/issues/43 я выкладывал конфиг для Home Assistant, который приемлемо работает с RK-M171S, у которого нет подсветки.
Добавил RK-G211S с подсветкой, добавил switch для включения подсветки.
Также убрал галочку Common Command/Response Topics чтобы были разные топики для команд и состояния.
```
climate:
  - platform: mqtt
    name: Kettle2
    retain: false
    min_temp: 0
    max_temp: 100
    temp_step: 5
    precision: 1
    availability_topic: "r4s/xxx/status"
    current_temperature_topic: "r4s/xxx/rsp/temp"
    temperature_state_topic: "r4s/xxx/rsp/heat_temp"
    temperature_command_topic: "r4s/xxx/cmd/heat_temp"
    mode_command_topic: "r4s/xxx/cmd/heat_temp"
    mode_command_template: >
      {% if value == "off" %}
      0
      {% elif value == "heat" %}
      85
      {% else %}
      100
      {% endif %}
    mode_state_topic: "r4s/xxx/rsp/json"
    mode_state_template: >
      {% if value_json.state == 0 %}
      off
      {% elif value_json.target > 0 %}
      heat
      {% else %}
      auto
      {% endif %}
    modes:
      - "off"
      - "heat"
      - "auto"

sensor:
  - platform: mqtt
    name: "kettle2_temp"
    state_topic: "r4s/xxx/rsp/temp"
    unit_of_measurement: '°C'

  - platform: mqtt
    name: "kettle2_heat_temp"
    unit_of_measurement: '°C'
    state_topic: "r4s/xxx/rsp/json"
    value_template: > 
      {% if value_json.target > 0 %}
        {{ value_json.target }}
      {% elif value_json.state != 0 %}
        100
      {% else %}
        0      
      {% endif %}

  - platform: mqtt
    name: Kettle2 Подсветка
    icon: mdi:lock
    state_topic: "r4s/xxx/nightlight"
    command_topic: "r4s/xxx/cmd/nightlight"
    retain: true
```

В UI добавил подсветку:
```
          - type: custom:simple-thermostat
            entity: climate.kettle2
            header:
              name: Чайник
              icon: mdi:kettle
              toggle:
                entity: switch.kettle2_podsvetka
                name: Подсветка
            step_size: 5
            control:
              hvac:
                "off":
                  name: Выкл.
                  icon: mdi:kettle-off
                heat:
                  name: 85°C
                  icon: mdi:kettle-alert
                auto:
                  name: Кипячение
                  icon: mdi:kettle-steam
            label:
              temperature: Сейчас в чайнике
```

В интеграции climate пришлось колхозить микс из состояний и температур, потому что всё свалено в кучу.
В template нельзя указать разные топики, можно указать топик для json и использовать его содержимое.
Режим выкл/подогрев/кипячение определяется вот так:
```
    mode_state_template: >
      {% if value_json.state == 0 %}
      off
      {% elif value_json.target > 0 %}
      heat
      {% else %}
      auto
      {% endif %}
```
И это прекрасно работало пока я не захотел включить подсветку.
При nightlight=ON вдруг `json/state=2` и climate показывает режим auto, то есть кипячение.
В template я не могу проверить /rsp/state=OFF
Проблема в том что и при кипячении json/state=2 тоже! То есть state в json неправильный! Не понятно почему он меняется при включении подсветки. И уж если меняется, то пусть при кипячении там будет что-то другое. Это нужно поправить!
А по хорошему это нужно переделать полностью, например как описано в #43.

alutov, 2021-07-22:
Строка Json   выводит в mqtt без изменений соответствующие поля состояния чайника.  Режим работы чайника определяет поле "prog". 0 - кипячение, 1 - нагрев, 3- подсветка. А поле "state" отражает только активирован этот режим (2) или нет(0). Или же чайник снят с подставки(4).

slydiman, 2021-07-22:
Интересно. 
Я ни разу не видел state=4, хорошо что есть status=OFFLINE (кстати почему большими буквами, а online маленькими?)
Про состояние json/prog в документации применительно к чайнику нет ни слова (только для рисоварки). 
Кстати, при включении подсветки чайника json/prog=3, json/state=2.
Если я выставлю r4s/xxx/cmd/prog=1 это включит подогрев? Или нужно ещё state=ON выставить?

Похоже идеальным решением было бы иметь возможность выставить r4s/xxx/cmd/json целиком, тогда любую логику можно было реализовать в templates HA.

Пока изменил конфиг в HA, посмотрим какие ещё открытия нас ждут
```
  - platform: mqtt
    name: Kettle2
    retain: false
    min_temp: 0
    max_temp: 100
    temp_step: 5
    precision: 1
    availability_topic: "r4s/xxx/status"
    current_temperature_topic: "r4s/xxx/rsp/temp"
    temperature_state_topic: "r4s/xxx/rsp/heat_temp"
    temperature_command_topic: "r4s/xxx/cmd/heat_temp"
    mode_command_topic: "r4s/xxx/cmd/heat_temp"
    mode_command_template: >
      {% if value == "off" %}
      0
      {% elif value == "heat" %}
      85
      {% else %}
      100
      {% endif %}
    mode_state_topic: "r4s/xxx/rsp/json"
    mode_state_template: >
      {% if value_json.state == 2 and value_json.prog == 0 %}
      auto
      {% elif value_json.state == 2 and value_json.prog == 1 %}
      heat
      {% else %}
      off
      {% endif %}
    modes:
      - "off"
      - "heat"
      - "auto"
```

alutov, 2021-09-24:
Строка Json выводит только соответствующие поля состояния чайника **_без каких-то модификаций_**. Управление не реализовано. Сделано было в основном в целях совместимости с [другим подобным проектом](https://github.com/olehs/r4sGate). Что-то менять в json  пока не планировал.

closed 2021-09-24 by alutov