alutov/ESP32-R4sGate-for-Redmond #48: Прошивка m5atom lite

rengaboy, 2021-09-13:
Доброго времени.
Попробовал прошить m5atom вашей прошивкой, прошивается успешно, но потом устройство не подаёт признаков жизни. К wifi не подключается, светодиоды не горят. Может ли это быть связано с нестандартной распиновкой, и если да, то сможете ли вы доработать прошивку для этого девайса, уж больно он удобный? Готов проспонсировать покупку для тестов.

alutov, 2021-09-16:
Честно говоря, не вижу причин, почему не работает прошивка, если m5atom собран на esp32. Если шили файлом fr4sGate.bin, то стоит попробовать файл fqr4sGate.bin, он с загрузчиком qio. Можно также глянуть лог при старте прошивки.

rengaboy, 2021-09-16:
Шил отдельно загрузчик, партишены и саму прошивку. Вечером попробую qio прошить, отпишусь тут

rengaboy, 2021-09-16:
`F:\ESP8266\r4sgate>esptool.py --chip esp32 --port COM6 --baud 750000 --before default_reset write_flash -z --flash_mode qio --flash_freq 80m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions.bin 0x10000 r4sGate.bin
esptool.py v3.1
Serial port COM6
Connecting.....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:a9:1b:70
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 750000
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x001a7fff...
Flash params set to 0x002f
Compressed 25040 bytes to 15481...
Wrote 25040 bytes (15481 compressed) at 0x00001000 in 0.6 seconds (effective 348.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 508.3 kbit/s)...
Hash of data verified.
Compressed 1669488 bytes to 963459...
Wrote 1669488 bytes (963459 compressed) at 0x00010000 in 17.7 seconds (effective 752.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

И абсолютная тишина. fqr4sGate.bin, аналогичная ситуация.

alutov, 2021-09-22:
По умолчанию экран теперь выключен. Прошивка должна работать. Как я понял из документации, в чипе ESP32-PICO-D4 для чтения флешки используются другие ноги, которые в прошивке как раз используются для управления экраном.

rengaboy, 2021-09-22:
Благодарю, все заработало.

alutov, 2021-09-24:
Отлично.

closed 2021-09-24 by alutov