alutov/ESP32-R4sGate-for-Redmond #52: Статистика

Drafteed, 2021-09-29:
Простите, что снова тревожу.. Просто перешел с кастомной интеграции для HA на вашу прошивку, настраиваюсь  :) 

В предыдущей версии кастомного компонента (которую я использовал) была статистика включений и количество потребленной энергии. [Тут можно глянуть как его получить](https://github.com/mavrikkk/ha_kettler/blob/217e5088ba75329b2b416b795fd73ba047db8b7e/custom_components/r4s_kettler/__init__.py#L242). Возможно это реализовать?

alutov, 2021-09-30:
Подумаю. И куда это все выводить?

Drafteed, 2021-09-30:
Отсылать в mqtt, как rssi :)

Drafteed, 2021-09-30:
В компоненте эти данные запрашивались с каждым обновлением (~15/30сек). Думаю чтобы не грузить эфир лишними запросами, ее можно запрашивать например раз в 15-30 минут.

Drafteed, 2021-10-01:
И еще добавлю сюда, чтобы отдельный не открывать:

При снятии чайника с подставки в HA постоянно вылетает ошибка:
```Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:480
Integration: MQTT (documentation, issues)
First occurred: 0:27:36 (18 occurrences)
Last logged: 14:30:10

Invalid modes mode: offline```

alutov, 2021-10-16:
Добавил статистику.

Drafteed, 2021-10-16:
Очень круто! Спасибо за фикс с режимом и новые переключатели. 
А с каким интервалом статистика обновляется? Уже два часа после обновления, включил его раз 5, но пока так и не прилетело )

[image]

Drafteed, 2021-10-16:
А хотя в брокере данные видно, что-то не так с дискавери.

[image]

alutov, 2021-10-17:
Да, есть такое. Я тестировал на совмещенных топиках. Поправил.

Drafteed, 2021-10-19:
Большое Вам спасибо! Теперь все просто замечательно :)

closed 2021-10-19 by Drafteed