alutov/ESP32-R4sGate-for-Redmond #56: Поддержка мультиварки RMC-M961S

Timoxa1329, 2021-11-14:
Будет ли добавлена поддержка мультиварки RMC-M961S?

alutov, 2021-11-15:
Можно попробовать подключить мультиварку в режиме 903. Если получится, тогда нужно только поправить список программ. Если же не получится, нужно снимать лог общения мультиварки и родного приложения.

Timoxa1329, 2021-11-17:
День добрый. Подключить получается и оно работает. Действительно необходимо только слегка подправить список программ. Премного благодарен за помощь

alutov, 2021-11-17:
Напишите, какие программы 903 соответствуют каким в 961. И желательно нужную температуру и время каждой программы 961.

Timoxa1329, 2021-11-18:
903 - 961
Multicooker - Рис/крупы, 10 мин, 100 °C
Milk_porridge - Жарка, 15 мин, 150 °C
Stewing - На пару, 25 мин, 100 °C
Frying - Выпечка, 1 час, 140 °C
Soup - Тушение, 1 час, 100 °C
Steam - Мультиповар, 30 мин, 100 °C
Pasta - Плов, 1 час, 110 °C
Slow_cooking - Суп, 1 час, 100 °C
Hot - Молочная каша, 30 мин, 100 °C
Baking - Йогурт, 8 часов, 40 °C

alutov, 2021-11-19:
Добавил, можно пробовать.

closed 2021-12-23 by alutov