alutov/ESP32-R4sGate-for-Redmond #58: смена домена на water heater

MaxWh1t3, 2021-12-04:
Не знал куда написать, простите. Зачем выбран домен климат, когда есть прекрасный water_heater? И например water_heater.kettle прекрасно себя в нем чувствует, по крайне мере интеграция от https://github.com/mavrikkk/ha_kettler туда его и определяет, что довольно удобно. Но в той интеграции чайник постоянно отваливается, пропускает команды и прочее, поэтому решил попробовать вашу и через 20 мин у меня еще больше выбор управления параметрами, даже энергию считает) Но не помешала бы более подробная статья по настройкам или хотя бы пояснения, т.к. не все очевидно...

alutov, 2021-12-05:
Нет пока в Mqtt Discovery сущности water_heater. Будет - переделаю, это быстро. Тем более, сам сейчас решил переехать на ХА, и потому заинтересован. По статье, как что смог уже написал, подробнее есть статьи, быстро ссылку не дать не смогу, нужно их поискать в инете.

MaxWh1t3, 2021-12-05:
спасибо!
про статью это к тому, что мне не пришлось всякие топики и прочее прописывать- все само пробросилось, очень удобно! ну и разбиение на абзацы не помешает, а то немного в кучу как-то.

closed 2021-12-23 by alutov

MaxWh1t3, 2021-12-24:
Спасибо! Но пока и так удобно вполне)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Без
вирусов. www.avast.ru
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>