alutov/ESP32-R4sGate-for-Redmond #60: Обновление прошивки

RavenHuginn, 2022-01-18:
Здравствуйте.
Подскажите есть настроенных шлюз, при обновлении через web настройки сохраняются или нужно будет заново настроить шлюз?

alutov, 2022-01-18:
Все сохраняется.

Drafteed, 2022-01-19:
Почему-то при обновлении через веб, браузер выкидывает CONNECTION_RESET.

alutov, 2022-01-20:
Обновление проверял на едже, хроме и файфоксе. Возможно, плохой уровень вайфая. Или же в есп старая версия, в которой при обновлении не закрываются все бле соединения.

Drafteed, 2022-01-20:
Очень интересно - с винды обновляется. В маке, в любом браузере (проверил Safari, FF, Chrome) нет.

alutov, 2022-01-20:
Как раз предсказуемо. Дело в том, что у разных браузеров приходящие на есп заголовки при обновлении немного отличаются. Задача отрезать заголовок и писать только данные. Где-то может что-то не учел ввиду отсутствия у меня информацмм.

Drafteed, 2022-01-20:
@alutov 

Вот что улетает вместе с файлом:

``` 
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Accept-Encoding: gzip, deflate
Accept-Language: ru,en-US;q=0.9,en;q=0.8,ru-RU;q=0.7,zh-TW;q=0.6,zh;q=0.5
Cache-Control: no-cache
Connection: keep-alive
Content-Length: 1587194
Content-Type: multipart/form-data; boundary=----WebKitFormBoundarye7P6hVQ3fslBtuMk
DNT: 1
Host: 192.168.1.11
Origin: http://192.168.1.11
Pragma: no-cache
Referer: http://192.168.1.11/update
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/97.0.4692.71 Safari/537.36
```

alutov, 2022-04-11:
Если есть возможность, загрузите последнюю версию, а потом запишите лог esp32 при попытке обновления из mac os. Там дожен записаться post заголовок.

alutov, 2022-04-28:
Поправил обновление для macos/ios.

closed 2022-04-30 by alutov