alutov/ESP32-R4sGate-for-Redmond #63: Конвектор REDMOND SkyHeat 7001S

Su4oNoK, 2022-02-13:
Можно ли добавить новое устройство конвектор REDMOND SkyHeat 7001S ? 
Сейчас пробую управлять им как SkyPort 103S, но во первых нет возможности выбирать режим мощности 25/50/75/100 
А во вторых, пока не нашел причину, но если выставить режим физической кнопкой на самом обогреватели, он отключается от ESP и висит там в оффлайн статусе, а потом становится Power RCH-7001S not connected.
Буду благодарен если поможете решить проблему.

alutov, 2022-02-14:
Чтобы добавить регулирование мощности, нужно снять лог блютуса в телефоне вовремя работы программы и выбора режимов. Что касается отключения, сказать ничего не могу, такого прибора не видел. Если же брать розетку, то там долгое нажатие на кнопку вводит розетку в режим сопряжения, а если продолжать держать кнопку, то в конце концов розетка забывает все подключения. Может что-то подобное происходит и с нагревателем.

Su4oNoK, 2022-02-14:
С подключением - зажал кнопку сопряжения секунд на 20, как я понял устройство всех забыло. Повторно прицепил к esp пока вроде норм, буду наблюдать. 

Вопрос со снятием лога - есть гайд как это правильно сделать?

alutov, 2022-02-16:
Лог уже прислали, конвектор RCH-7001S добавлен. Возможно, он совместим и с другими конвекторами редмонда, в частности с RCH-4529S.

artt652, 2022-02-16:
Точно будут работать конвекторы RCH-7002S, RCH-7003S, платы управления у них с RCH-7001S одинаковые, различия только в длине и мощности нагревательного элемента.

alutov, 2022-02-16:
Ага, спасибо, отмечу это в описании.

Su4oNoK, 2022-02-16:
Всё работает, огромное спасибо!

closed 2022-02-16 by Su4oNoK