alutov/ESP32-R4sGate-for-Redmond #66: топики SkyKettle RK-GхххS

Svet98, 2022-04-11:
Андрей, здравствуйте!

Подскажите, для эл.чайников Вы случайно не расширяли функционал?
описание на лицевой страничке команд MQTT у Вас до последних обновлений.
интересуют, если есть, команды на:
- время кипячения (Smart Boil)
- режим кипячения с переходом в подогрев
- режим подогрева на указанное время
- настройка RGB на температуру
- более тонкая настройка ночника, подсветки (яркость, интервал переключения оттенков)
- блокировка ручного управления

Спасибо!

Svet98, 2022-04-11:
не знаю, поможет Вам, в Вашем проекте,
нашла проект на питоне для линукса, управление редмонд чайником
https://github.com/ClusterM/skykettle-ha

alutov, 2022-04-11:
Пока не делал ничего из указанного кроме 2 пункта. Не знаюРежим кипячения с переходом в подогрев включается, если в режиме подогрева включить кипячение.

Svet98, 2022-04-11:
Спасибо!)

Svet98, 2022-04-27:
Андрей, добрый день!
Может имеет смысл открыть раздел вики и поместить туда MQTT топики?
Разделить по устройствам и по мере развития Вашего проекта можно обновлять.
Я немного структурировала Вами описанные для чайника + добавила до чего сама дошла)

[Mqtt топики.txt](https://github.com/alutov/ESP32-R4sGate-for-Redmond/files/8570008/Mqtt.txt)

Спасибо)

artt652, 2023-12-23:
Для информации, актуальные топики для чайника:

[image]

sviperz, 2024-10-11:
Не совсем понимаю логику работы `heat_temp`. Основной мой сценарий это нагрев (не кипячение) до разных значений, обычно до 85.

Для этого в интеграции брокера WQTT.RU с умным домом Яндекса, на кнопку включения прописано не `boil`, а `heat`:

[image]

Регулятор температуры завязан на `heat_temp`:

[image]

Задаю температуру - сетается  `heat_temp`, нажимаю кнопку "Включить" - отправляется команда `heat`. Но как только я это делаю, сразу применяется `heat_temp`, который был до этого выставлен в веб-интерфейсе шлюза и нагрев идет до этой температуры из шлюза. Это видно по логам WebSocket клиента:

[image]

[image]

[image]

Почему так происходит и как сделать, чтобы использовалась именно температура, которая задана через `heat_temp`, а не использовалась последний раз через интерфейс шлюза?

Svet98, 2024-10-12:
у меня несколько режимов подогрева.
- выполнение последней установленной температуры
- выбор температуры
последняя работает следующим образом:
отправка на heat_temp нужной температуры.
heat_temp - это подогрев до указанной в этой переменной значения.

alutov, 2024-10-12:
Установка heat_temp сразу же включает режим подогрева. При выключении режима подогрева heat_temp сбрасывается в 0. В старых чайниках типа 173 иначе не получается. Шлюз запоминает последнее значение heat_temp и восстанавливает его при включении режима подогрева топиком heat.

Svet98, 2024-10-12:
Для олдовых прошивок

r4s0/devaddr/cmd/boil <-- 0/off/false - выключение кипячения, 1/on/true - включение кипячения. Если перед этим был включен подогрев, включается кипячение с последующим подогревом;
r4s0/devaddr/cmd/heat <-- 0/off/false - выключение подогрева, 1/on/true - включение подогрева с последней запомненной шлюзом температурой. При старте шлюза температура берется из поля Heat в настройках;
r4s0/devaddr/cmd/heat_temp <-- 30...90 - включение подогрева, > 97 - выключение, < 30 - выключение, если подогрев был включен, последняя температура если был выключен;
r4s0/devaddr/cmd/boiltime <-- -5...5 - время кипения чайника;

sviperz, 2024-10-12:
@Svet98, @alutov, спасибо за подробности, теперь логика ясна.

Проблема, видимо, в моем брокере WQTT.RU. Когда напрямую отправляю в топик `r4s0/devaddr/cmd/heat_temp` значение температуры, то в логе видно, что сообщение отправляется, но при этом больше ничего не происходит - чайник никак не реагирует:

[image]

Соответственно, шлюз не запоминает это значение температуры, и когда отправляю `r4s0/devaddr/cmd/heat = ON`, то шлюз просто использует последнее значение, которое он знает.

Пока не знаю, что с этим делать... У WQTT.RU удобная интеграция с УДЯ, но выходит, что не работает так, как мне нужно. Попробую перейти на свой брокер в Home Assistant, с MQTT до этого шлюза не встречался, буду разбираться.

alutov, 2024-10-13:
Можно попробовать использовать совмещеннфе топики (по памяти common command response topics). То есть для команд и состояния одни и те же топики. Будет гарантия, что команды попадают куда нужно. И еще одна тонкость. Числа в топиках данных идут как ascii текст. Может в этом дело.

sviperz, 2024-10-13:
Все оказалось гораздо проще - моя ошибка в указании топика для `heat_temp`. Написал `r4s` вместо `r4s0`)

Прошу прощения за шумиху. Все работает! И с раздельными топиками и с совмещенными. Еще раз спасибо!

alutov, 2024-10-13:
Да, раньше шлюз с номером 0 писал в топик r4s. Но при появлении в шлюзе опции BLE Monitor топик r4s используется BLE Monitor, а шлюз с номером 0 пишет в топик r4s0.

ZhdanOFF90, 2025-09-24:
Здравствуйте, на чайнике RK-G212S (Name: RFS-KKL002, Ver: 2.59) в JSON топик "state":{"temp":94,"target":0,"state":0,"prog":0,"beep":1,"boil":0,"volume":["?",0.0],"rgb":[0,0,100]}} при включении boil не передается "1" соответственно нет возможности определить включено кипячение или нет.