alutov/ESP32-R4sGate-for-Redmond #68: ведения проекта

Raerten, 2022-04-22:
Большое пожелание - вести проект с правильной работой с гитом: 
- не удалять/создавать файл с кодом - теряется история, сильно затрудняет просмотр изменений, затрудняет ведение проекта несколькими людьми
- заливать бинарники в релиз гитхаба, а не комитить в проет (сильно разрастается репозиторий)
- завести отдельную ветку с разработкой (dev), не заливать обновления в мастер с одинаковым номером релиза

alutov, 2022-04-22:
Вот даже пока не знаю, что сказать. Не храню на гите ничего. И раньше не доверял, а сейчас тем более. Уже приходило письмо о том, что нахожусь в подсанкционном регионе. Так что рубануть доступ могут на раз. Есть и еще причина. Есть русские буквы в меню мультиварки. Их тупо вбил в ascii, и, если в винде стоит выбор отображение не юникода как русских букв, все срабатывает правильно. Но гит русские буквы корежит, и потом скачать исходник я уже не могу. Потому и заливаю его каждый раз.

alutov, 2022-04-22:
Уже смотрел, пока сильно не вникал, изучу еще.

Raerten, 2022-04-22:
вообще у меня уже есть желание немного перелопатить исходники- попилить на несколько файлов, форматирование в порядок привести. и мне какжется что проверка на дефайн отключение экрана не всегда в нужных местах - у меня без него не собралось

Raerten, 2022-04-22:
и да, спасибо огромное, подключил на неделе 3 девайса, все работают

Svet98, 2022-04-25:
желательно и файл #readm описывать с версиями в изменениях. 
Текущий трудно читаем, не понятно что в каком релизе было сделано.
Особенно касается когда используется частичный функционал, например без монитора.

П.С,
Ещё раз огромное спасибо за Ваш проект!
П.П.С.
на днях закончила интеграцию в MajorDoMo. Как и у всех, с временем беда)) как только закончу ролик с инструкцией, могу скинуть линк. Инструкцию делаю для форумчан. Сама люблю много вопросов задавать))

bvp, 2022-06-05:
Тоже есть несколько предложений по ведению проекта:
1. Использовать clang-format для форматирования кода - можно накидать конфиг в [конфигураторе](https://zed0.co.uk/clang-format-configurator/)
2. Разбиение на более мелкие файлы, а не одну "простыню" кода (хотя сам этим часто грешу, но стараюсь исправляться)
3. Возможность сборки проекта с помощью [Platformio](https://platformio.org/) - есть рабочий конфиг

alutov, 2022-06-06:
Первый пункт поизучаю. Что касается остального. Я вообще по си не специалист, мне больше нравится ассемблер. Там все просто, все метки и процедуры можно размещать как угодно, для этого есть 2 прохода. Здесь же нужно следить, чтобы вызов процедуры был позже ее декларирования. Разбирая похожий проект [olehs](https://github.com/olehs/r4sGate) на ардуино с кучей файлов и перекрестных ссылок, просто начинал путаться, что где и за чем. Потому стараюсь без надобности не делить файлы. Это нехорошо,  конечно, надо исправляться.) Так вот, на ардуино проект так и не пошел, вероятно из-за кривизны ble библиотеки, хотя оперативку жрал хорошо. Перейдя на чистый esp-idf и написал недостающее с нуля, получил и больше оперативки, и лучшую стабильность, хотя на доводку при работе с несколькими соединениями ушел еще год.  С тех пор скептически отношусь и к ардуино, и к Platformio(это тоже по сути ардуино?) и к esphome. Последний, правда, переползает на esp-idf, но говорят, не полностью. За Platformio ничего не знаю. Если в нем можно собрать и работает нормально, выложите, а я сделаю ссылочку.

alutov, 2022-06-08:
Лутов Андрей) Спасибо. Надо поправить описание как-нибудь, сделаю и ссылочки на эти ресурсы.)

Svet98, 2022-06-08:
Извините... Поправила. Спасибо)

ggagnidze, 2023-04-16:
А может с веб-мордой что-нибудь сделаем? Выглядит не прям современно. Может помощь какая-то нужна с этим?

alutov, 2023-04-16:
Ну совершенствовать можно бесконечно). Может и нужна помощь, сам не особо вникал в html программирование. Но уже сейчас осталось где-то 11% и 9% свободного места для esp32 и esp32c3 с 4M флешкой. И не сильно хочется нагружать есп лишней работой.

olgasafo, 2023-09-11:
Благодарю вас за такой проект.
Я подключала приводы AM43.
Все рботает, но ввиду видимо частых запросов со стороны гейта, батарея приводов разряжается очень быстро.
Можно ли попросить вывести настройку частоты запросов в интерфейс?
Или может есть какой-то иной способ увеличить жизнь батарей?
Так же не все настройки понятны, может быть будет время сделать короткий faq по веб интерфейсу?

alutov, 2023-09-12:
Интервал запроса шлюзом состояния привода штор три минуты. В папке jpg вариант с интервалом 10 минут. Уж не знаю, поможет ли. Вообще привод идет с солнечной батареей. Правда у меня она через окно почему-то аккумулятор практически не заряжает. Взял батарею тоже на 12 вольт, но побольше, вроде бы стало лучше.

olgasafo, 2023-09-12:
Благодарю, попробую. Но вы говорите о интервале 10 минут. А я в сообщениях mqtt вижу данные о уровне сигнала передатчика привода с интервалом в 3-8 секунды. Это же опрос? Возможно в этом проблема быстрого расхода?

alutov, 2023-09-12:
Передача может быть и по инициативе привода. При изменении позиции штор или освещенности, или разряде батареи. У меня при заряде в интервале 96-100% привод постоянно шлет нотификационные сообщения. Потом вроде успокаивается.

olgasafo, 2023-09-22:
>  В папке jpg вариант с интервалом 10 минут. Уж не знаю, поможет ли.

Двухнедельный эксперимент с прошивкой с 3х минутным запросом и 10 минутным показали абсолютно идентичный разряд. Так что дело не в опросе устойства, а видимо в непрерывном блютус соединении.
Приводов подключенных через r4sGate хватает на 14-17 дней на стоковой батареей (с износом в 3 года), в отличие от работы стационарно (около 30-40 дней).

> Вообще привод идет с солнечной батареей.

Имхо вообще не работает никак... Или солнца мало, или батарея фыфел. В один из дней даже показалось, что разряд из за неее увеличился. :-)

Вот такая вот печальная статистика.

alutov, 2023-09-24:
Кто-то мне писал, что родная солнечная батарея заряжает фккумулятор, если ее ставить до оконного стекла, то есть на улице. Мне это никак не подходило, пришлось поставить [батарею помощнее](https://aliexpress.ru/item/1005004905669549.html?spm=a2g2w.orderdetail.0.0.6f384aa6HabEEF&sku_id=12000031151630435). Пока ттестирую.

Niorix, 2023-10-08:
Коли тут подобие чата образовалась залезу тоже. Пару лет назад был запрос на изменение домена на Water Heater, но был отклонен т.к. нет поддержки на уровне mqtt. С тех пор ничего не поменялось? https://www.home-assistant.io/integrations/water_heater.mqtt/ не оно?

alutov, 2023-10-08:
Вроде бы оно. Только я не помню, что там станет лучше, если переделать на water heater?

Niorix, 2023-10-09:
> Вроде бы оно. Только я не помню, что там станет лучше, если переделать на water heater?
До этого пользовался интеграцией SkyKettle. Но с версии 2023.9.x через ble_Proxy интеграция перестала корректно работать, а напрямую не домтает. Там не используется mqtt и чайник представляет из себя  Water Heater. Из плюсов для себя заметил отсутствие "непонятных" для чайников режима охлаждения и авторежима. Есть режимы кипячения, подогрева, кипечения + последующего подогрева. Хотелось такого же...

> Посмотрел детальнее, что-то не впечатляет.

Да уж, похоже лучше как есть оставить.

alutov, 2024-02-23:
Все же добавил water heater для чайников.