alutov/ESP32-R4sGate-for-Redmond #85: Нет подключения к брокеру MQTT

KaportsevIA, 2022-06-06:
Доброго времени суток.
У меня возникла такая проблема.
r4s не подключается к брокеру MQTT

Брокер развернут на сервере ХА в докере. Стоит авторизация на брокере.
В статусе r4s gateway - 0/disconnected.
соответственно топики в моските не создаются.

[image]
[image]

alutov, 2022-06-06:
Судя по картинке, в шлюзе не прописан пароль, так как нет точек. Возможно, в пароле есть особенный символ, который не дает ему прописаться.

KaportsevIA, 2022-06-06:
Осталось только понять, как это все совместить в одну карточку.

Raerten, 2022-06-07:
> Осталось только понять, как это все совместить в одну карточку

например https://github.com/benct/lovelace-multiple-entity-row

alutov, 2022-06-07:
Мелькает мысль отказаться от климата, и сделать температуру на input и свичах. light правда придется оставить, лучше все равно в дискавери не придумать.

KaportsevIA, 2022-06-08:
Кстати для объединения сущностей чайников было бы правильнее транслировать их в MQTT как

water_heater.kettlexxxx_heat
water_heater.kettlexxxx_boil
water_heater.kettlexxxx_light
water_heater.kettlexxxx_lightswitch
water_heater.kettlexxxx_temp
water_heater.kettlexxxx_volume
water_heater.kettlexxxx_volumelast
water_heater.kettlexxxx_energy
water_heater.kettlexxxx_ontime
water_heater.kettlexxxx_count

Тогда они будут попадать под класс - водонагреватель

alutov, 2022-06-08:
Это где-то можно прочитать подробнее? Было бы замечательно, ибо насколько я знаю, по мктт дискавери такого класса пока нет, есть сенсоры, инпуты, климат и лайт. все это впихиваю в один девайс, все равно он нагревателем или климатом не становится(.

KaportsevIA, 2022-06-08:
К сожалению не могу сейчас посмотреть в ХА (с 9-00 дома нет света). Но поидее, это да ХА-шная приблуда. Т.к. когда чайник был через редмондовский шлюз RSC-11S подключен к Алисе, Алиска передавала его в ХА как класс устройство water_heater. Соответственно, он мог кипятить воду, выставлять температуру (простейшие действия, что больше и не надо).

https://www.home-assistant.io/integrations/water_heater/
Либо сделать возможность смены того как он именуется сейчас, чтото типа r4s0.<имя из настроек типа G200S>. Это скорее всего берется из имени топика MQTT r4sGate0 например.
По-крайней мере чайник имею, могу попроверять.

alutov, 2022-06-08:
Ок, я попробую сделать на днях вот этот первый вариант в тестовой сборке и посмотрим, а вдруг прокатит) Это по идее в идентификаторах надо делать, имя вроде меняется из интерфейса, идентификатор нет.

KaportsevIA, 2022-06-08:
Я кстати видел там опцию Hass Discovery но что разницу в ней между Вкл/Выкл не заметил.

alutov, 2022-06-08:
Если она была вкл, то конфигурация прописалась и сталась в брокере. Можно и выключить. Чтобы ее убрать, нужно выбрать в настройках Delete Mqtt topics. Почистятся все топики, и устройства пропадут из ХА.

KaportsevIA, 2022-06-15:
Кстати заметил отвалы шлюза из сети. Периодичность точно не скажу (Чайником пользуемся раза 2-3 в день). Но сам шлюз просто пропадает из сети, хотя питание на него подается. Вряд-ли это неисправность платы (ESP32DEVKIT V1), т.к. у меня их несколько, и пропадают обе, которые прошиты под R4SGate. Т.е. их не видит роутер вообще. Щаз поднял на основном KeenViva вторую гостевую сеть с логином и паролем r4s/12345678 понаблюдаю.

KaportsevIA, 2022-06-15:
Кстати подскажи, плиз, для чего вот эти строки
[image]
?

alutov, 2022-06-16:
Может быть слабый сигнал wifi  у есп. Ппо моим наблюдениям, при -80 примерно начинаются проблемы. Может неудачно стоят каналы в роутере. Оба стандарта ведь делят одни частоты. У меня 1, 6, 11 каналы по 20МГц. А может просто не выдерживает роутер. Имею кинетик, но когда приезжают гости, начинаются отвалы. Вряд ли дело в прошивке, тут все стандартно. Строки это топики из мктт для отображения на экране шлюза напряжения, тока, температур и состояния котла. Есть в описании.

closed 2022-07-09 by alutov