alutov/ESP32-R4sGate-for-Redmond #86: Возможно ли изменить температурный интервал измерения оставшегося количества воды

Bajkot, 2022-06-07:
Произвожу нагрев исключительно до 90 градусов, завариваю кофе. Температура доходит до 90 градусов, но показания volume и volumelast не изменяются.

Хотелось бы спросить, возможно ли изменение интервала измерения например на 60-80 градусов.

alutov, 2022-06-08:
Сильно сдвигать вниз тоже не хочется, пропадут показания, если нагревать с неостывшую еще воду. Сдвинул вниз на 5 градусов. Сейчас 65-85.

Bajkot, 2022-06-08:
Параметры volume и volumelast меняются только при включении режима кипячения. При включении режима обогрева, в моем случае до 90 гр., не меняются показания volume и volumelast, а сбрасываются на ноль.

alutov, 2022-06-09:
Ну вообще-то только кипячение всегда и было. Сделал и нагрев, если установлена температура 85 и выше.

Bajkot, 2022-06-09:
Благодарю за добавление функции измерения при обогреве и снижение на 65-85. Теперь могу полноценно реализовать функционал чайника. Спасибо и от моего товарища. Пользуемся плодами проекта с самого начала.