alutov/ESP32-R4sGate-for-Redmond #90: ESP32-C3

DmytroMelnyk, 2022-06-28:
Подскажите, пожалуйста, а возможно сделать билд под ESP32-C3?

alutov, 2022-06-28:
Скажу честно, не вникал, чем она отличается от обычной. Знаю только, что на атоме прошивка работает, а там ESP32-PICO-D4.

DmytroMelnyk, 2022-06-28:
In file included from main/r4sGate.c:347:
main/tft/tft.c: In function 'lcd_init':
main/tft/tft.c:1197:23: error: 'LEDC_HIGH_SPEED_MODE' undeclared (first use in this function); did you mean 'LEDC_LOW_SPEED_MODE'?
         .speed_mode = LEDC_HIGH_SPEED_MODE,   // timer mode
                       ^~~~~~~~~~~~~~~~~~~~
                       LEDC_LOW_SPEED_MODE
main/tft/tft.c:1197:23: note: each undeclared identifier is reported only once for each function it appears in
In file included from main/r4sGate.c:347:
main/tft/tft.c: In function 'tftinit':
main/tft/tft.c:7:21: error: 'HSPI_HOST' undeclared (first use in this function); did you mean 'SPI3_HOST'?
 #define SPI_PORT    HSPI_HOST
                     ^~~~~~~~~
main/tft/tft.c:1331:59: note: in expansion of macro 'SPI_PORT'
  if (PIN_NUM_MISO == PIN_NUM_MOSI) ret=spi_bus_initialize(SPI_PORT, &buscfg, 0);
                                                           ^~~~~~~~
main/r4sGate.c: In function 'hw_timer_callback':
main/r4sGate.c:354:10: error: 'timg_dev_t' {aka 'volatile struct timg_dev_s'} has no member named 'int_clr_timers'; did you mean 'int_clr'?
  TIMERG0.int_clr_timers.t0 = 1;
          ^~~~~~~~~~~~~~
          int_clr
main/r4sGate.c: In function 'mqtt_event_handler_cb':
main/r4sGate.c:10349:16: error: 'LEDC_HIGH_SPEED_MODE' undeclared (first use in this function); did you mean 'LEDC_LOW_SPEED_MODE'?
  ledc_set_duty(LEDC_HIGH_SPEED_MODE, LEDC_CHANNEL_0, duty);
                ^~~~~~~~~~~~~~~~~~~~
                LEDC_LOW_SPEED_MODE
Compiling .pio\build\esp32dev\asio\asio\asio\src\asio.o
Archiving .pio\build\esp32dev\esp-idf\app_update\libapp_update.a
Archiving .pio\build\esp32dev\esp-idf\app_trace\libapp_trace.a
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_common.o
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_common_loader.o
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_clock_init.o
Indexing .pio\build\esp32dev\esp-idf\app_update\libapp_update.a
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_flash.o
Indexing .pio\build\esp32dev\esp-idf\app_trace\libapp_trace.a
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_mem.o
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_random.o
*** [.pio\build\esp32dev\main\r4sGate.o] Error 1

DmytroMelnyk, 2022-06-28:
попробовал сбилдить в PlatformIO... вот такие вот ошибки выскочили

DmytroMelnyk, 2022-06-28:
tft monitor я могу закоментить - он мне не нужен... но вот что делать с ошибками компиляции в r4sGate.c я не знаю :(

alutov, 2022-06-28:
Можно попробовать поменять на то, что он предлагает. Наверное это обрезанная версия обычной есп32.

DmytroMelnyk, 2022-06-29:
Да в режиме обычной esp, т.е. без 
```
board_build.mcu = esp32c3
board_build.variant = esp32c
```
собирается без ошибок

DmytroMelnyk, 2022-06-29:
Мне по сути нужно управление только чайником, а не весь ГейтВей... Если бы Вы подсказали что можно вырезать оттуда :). Я пробовал комментить
```
#include "tft/tft.c"
```
ошибок стало больше и совсем разных :(

DmytroMelnyk, 2022-06-29:
Я просмотрел исходный код и не нашел упоминания о чайнике RK-M216S. Подскажите, пожалуйста, он поддерживается?

Raerten, 2022-06-29:
> Я просмотрел исходный код и не нашел упоминания о чайнике RK-M216S. Подскажите, пожалуйста, он поддерживается?

у меня так заработал 
[image]

alutov, 2022-06-29:
В следующей версии будет возможность отключения экрана при сборке. Пока не могу выложить, так как никак не могу проверить одно устройство.

DmytroMelnyk, 2022-07-03:
Добрый вечер! Попробовал я использовать рекомендации IDE, т.е. заменил:
```
LEDC_HIGH_SPEED_MODE => LEDC_LOW_SPEED_MODE
HSPI_HOST => SPI3_HOST
TIMERG0.int_clr_timers.t0 => TIMERG0.int_clr_timers.t0_int_clr
TIMERG0.hw_timer[0].config.alarm_en => TIMERG0.hw_timer[0].config.tx_alarm_en
```

так же добавил в platformio.ini:
```
build_flags = -DBLE_42_FEATURE_SUPPORT=TRUE -DCONFIG_BT_BLE_42_FEATURES_SUPPORTED=TRUE
```

Проект скомпилировался, я даже сумел его залить в ESP32C3. Плата подключилась к гостевому Wi-Fi, получила IP адресс, но вот к веб-серверу я уже подключится не смог. Зато получил логи:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (31) boot: ESP-IDF 4.4.1 2nd stage bootloader
I (31) boot: compile time 16:32:43
I (31) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32c3: SPI Speed      : 40MHz
I (45) boot.esp32c3: SPI Mode       : DIO
I (49) boot.esp32c3: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (85) boot:  2 otadata          OTA data         01 00 0000e000 00002000
I (93) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (100) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (108) boot:  5 spiffs           Unknown data     01 82 003d0000 00030000
I (115) boot: End of partition table
I (119) boot_comm: chip revision: 3, min. application chip revision: 0
I (127) esp_image: segment 0: paddr=00010020 vaddr=3c160020 size=627c0h (403392) map
I (222) esp_image: segment 1: paddr=000727e8 vaddr=3fc91600 size=04f38h ( 20280) load
I (227) esp_image: segment 2: paddr=00077728 vaddr=40380000 size=088f0h ( 35056) load
I (236) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=158174h (1409396) map
I (540) esp_image: segment 4: paddr=001d819c vaddr=403888f0 size=08c48h ( 35912) load
I (549) esp_image: segment 5: paddr=001e0dec vaddr=50000010 size=00010h (    16) load
I (556) boot: Loaded app from partition at offset 0x10000
I (556) boot: Disabling RNG early entropy source...
I (570) cpu_start: Pro cpu up.
I (578) cpu_start: Pro cpu start user code
I (578) cpu_start: cpu freq: 160000000
I (578) cpu_start: Application information:
I (581) cpu_start: Project name:     r4sGate
I (586) cpu_start: App version:      1
I (590) cpu_start: Compile time:     Jul  3 2022 16:28:09
I (597) cpu_start: ELF file SHA256:  48440bdb5164096f...
I (602) cpu_start: ESP-IDF:          4.4.1
I (607) heap_init: Initializing. RAM available for dynamic allocation:
I (615) heap_init: At 3FCA6F20 len 000190E0 (100 KiB): DRAM
I (621) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (628) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (635) spi_flash: detected chip: generic
I (639) spi_flash: flash io: dio
I (643) sleep: Configure to isolate all GPIO pins in sleep state
I (649) sleep: Enable automatic switching of GPIO sleep configuration
I (657) coexist: coexist rom version 9387209
I (662) cpu_start: Starting scheduler.
Starting r4sGate...
[... 40 lines cut ...]
I (935) wifi:connected with r4s, aid = 2, channel 11, BW20, bssid = 34:97:f6:68:9e:79
I (935) wifi:security: WPA2-PSK, phy: bgn, rssi: -41
I (935) wifi:pm start, type: 1

I (935) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
W (955) wifi:<ba-add>idx:0 (ifx:0, 34:97:f6:68:9e:79), tid:0, ssn:0, winSize:64
I (1015) wifi:BcnInt:102400, DTIM:1
I (1695) esp_netif_handlers: sta ip: 192.168.1.223, mask: 255.255.255.0, gw: 192.168.1.1
I (1695) R4S: got ip:192.168.1.223
I (1695) R4S: Connected to ap SSID:'r4s' password:'12345678'
W (1705) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (1705) BTDM_INIT: BT controller compile version [d913766]
I (1715) BTDM_INIT: Bluetooth MAC: 84:f7:03:a6:64:02

I (1725) R4S: app_main init bluetooth

I (1745) R4S: Basic Auth string:
I (1745) R4S: esp32 BLE MAC Address:
I (1745) R4S: 84 f7 03 a6 64 02
I (1755) R4S: Init end free heap: 94148
```
P.S. Еще не знаю зачем, но на всякий случай проапдейтил версию espressif:
```
platform = espressif32@5.0.0
```

DmytroMelnyk, 2022-07-03:
я раскоментил пару логов, там где был ВебСервер и получил вот такие вот логи:
```
I (1755) R4S: Starting server on port: '80'
I (1755) R4S: Max URI handlers: '16'
I (1755) R4S: Max Open Sessions: '7'
I (1765) R4S: Max Header Length: '512'
I (1765) R4S: Max URI Length: '512'
I (1765) R4S: Max Stack Size: '20480'
I (1775) R4S: Init end free heap: 94164
```

DmytroMelnyk, 2022-07-03:
как можно добавить логирование к хендлерам?
httpd_register_uri_handler

DmytroMelnyk, 2022-07-03:
[image]
Добавил вот так вот лог, но при переходе по IP адресу в браузерной строке, лог этот не отобразился, но и страница не отрисовалась. Может действительно памяти мало :(

alutov, 2022-07-13:
Обновил версию, ее можно собрать без опции экрана. Вроде бы убрал по максимуму. Можно пробовать собрать для C3.

DmytroMelnyk, 2022-07-13:
Попробовал новую версию. Но все таки проблема в http сервере похоже. Заметил еще вот такой вот лог, это критично?
`I (1724) R4S: reg app failed, app_id 0004, status 128`
Если да, то может тут нужно severity warning

alutov, 2022-07-13:
Похоже, что не хватает памяти для запуска http сервера. Что в общем-то и  понятно, ведь разница в оперативке между обычной esp и esp-c3 (120к)  больше свободного озу в шлюзе, это где-то 85к при отсутствии BLE устройств в настройках.

alutov, 2022-07-13:
Ну правильно, он пытается взять память, а памяти не хватает.

DmytroMelnyk, 2022-07-20:
Оказывается что все работает. Проблема была на моей стороне. Когда я создал гостевую сеть, я выключил доступ с гостевой в мою локальную. Т.е. IP нового устройства я видел, а доступа к нему не имел :(. Работает все на ESP32C3)))

alutov, 2022-07-20:
Работает собранная юез экрана? И сколько свободной памяти показывает?

DmytroMelnyk, 2022-07-20:
Да. Экран я выключил. Вот что по памяти:
[image]

alutov, 2022-07-20:
Памяти больше даже, чем без экрана на обычной, там 98к кажется было. Она  бы и экран потянула, наверное, да выводов маловато.

DmytroMelnyk, 2022-07-20:
Выводов меньше. Но стоила она дешевле чем обычная :)

DmytroMelnyk, 2022-07-22:
Если можно, то было бы неплохо добавить что то вот такое вот в исходники
`//Reset irq and set for next time
#ifdef CONFIG_IDF_TARGET_ESP32C3
	TIMERG0.int_clr_timers.t0_int_clr = 1;
	TIMERG0.hw_timer[0].config.tx_alarm_en = 1;
#else
	TIMERG0.int_clr_timers.t0 = 1;
	TIMERG0.hw_timer[0].config.alarm_en = 1;
#endif
//`

Ну а затем я отключал TFT:
`
//#define USE_TFT
`
апгрейдил версию платформы в platformio.ini и добавлял сюда же модификацю esp32:
`
platform = espressif32@5.0.0
board_build.mcu = esp32c3
board_build.variant = esp32c3
`
билдил и заливал. Как по мне прозе всего это делать в VSCode + экстеншин "Platformio IDE"

alutov, 2022-07-25:
Добавил в исходники. При выборе С3 опция экрана отключается. Там нужно еще разбираться с SPI, пока не вникал, в чем отличия от обычной есп32.

DmytroMelnyk, 2022-07-25:
Точно не скажу, но когда собирал с экраном кажись оно на что то ругалось связанное с SPI. То ли именование констант было какое-то другое - уже не помню.

closed 2022-07-25 by DmytroMelnyk