alutov/ESP32-R4sGate-for-Redmond #96: MQTT ssl

utya, 2022-07-18:
Привет. Подскажите будет поддержка ssl/tls mqtt?

alutov, 2022-07-18:
Пока не думал, а нужно? Для локального брокера вроде не критично.

utya, 2022-07-18:
Да вот поднял в облаке настроил tls хотел всех уже туда перевести, но потом понял лучше наверно поднять локальный и забриджевать с внешним

alutov, 2022-07-22:
Немного почитал об mqtt с ssl. Честно говоря, не очень хочется возиться с сертификатами. Но. Есть в москито вариант использовать PSK, и даже вариант с  авторизацией и шифрованием  PSK. Очень удобно, как по мне. Но вот поддержки этого в HA не нашел.

utya, 2022-07-22:
> Немного почитал об mqtt с ssl. Честно говоря, не очень хочется возиться с сертификатами. Но. Есть в москито вариант использовать PSK, и даже вариант с авторизацией и шифрованием PSK. Очень удобно, как по мне. Но вот поддержки этого в HA не нашел.

ну не знаю, может будет полезно как у тасмота, они там двумя путями юзают, fingerprint и сашивают letsecrypt корневой сертификат

utya, 2022-07-22:
https://tasmota.github.io/docs/TLS/

alutov, 2022-07-22:
Ага, почитаю, спасибо.

alutov, 2022-07-25:
Добавил ssl по стандартной схеме. Пробовал на локальном брокере и самоподисанными сертификатами.

utya, 2022-07-25:
скинь пример, как пробовать, я проверю

closed 2022-07-25 by utya

utya, 2022-07-25:
> Похоже, что Mqtt Explorer имя сервера не проверяет
mqtt старые сертификаты, поэтому у него там чёто сломано, они ждут обновления electron

alutov, 2022-07-25:
Я сгенерил сертификаты на имя homeassistant. У меня имя хоста HA именно такое. Вбил сертификат в эксплорер, все нормально, проверяю из есп - ошибка проверки сертификата. День не мог понять, почему так , потом нашел опцию отключения проверки имени хоста - все стало нормально. Я думаю, что имя хоста это то, что пишется в поле сервера мктт или же полная строка имя сервера : порт. Сам не проверял.