alutov/ESP32-R4sGate-for-Redmond #101: Viomi Smart Kettle "V-SK152(Int)"

PilgrimF, 2022-07-26:
На прошивке 2022.07.25 перестал коннектиться Viomi Smart Kettle "V-SK152(Int)".
Откатился на 2022.06.03 - всё работает.

alutov, 2022-07-26:
Тяжело сразу что-то сказать, очень много изменений было. Если не коннетктится, на чем он останавливается? На этом:
Connecting MiKettle (Auth/Id:1116)?
Чайник в той же позиции и тот же номер шлюза? Use MAC in BLE Authentication не менялся? Да и вообще, настройки все бы проверить, могло что-то слететь. Если новая версия запишет, старая уже часть параметров прочитает. Если совсем глухо, нужно смотреть лог. У меня чайник  обычный MiKettle работает, от виоми он отличается только айдишником 275 против 1116. Есть еще режим MiUnknown, шлюз перебирает айдишники от 1 до 10000. Правда это долго.

PilgrimF, 2022-07-26:
Подключение останавливается на Connecting MiKettle (Auth/Id:1116).
Потом эта надпись пропадает и по новому кругу.
Ничего не менялось. Все параметры и еспшка те же, что и раньше.
Сейчас лог не вышлю, уехал из дома на несколько дней.
Попробую выслать лог в пятницу 29.07, если получится.

alutov, 2022-07-27:
Посмотрим лог, конечно. Думаю, проблема решится, если все же выполнить Format NVS, после чего, если чайник не подключится, перепривязать его снова. Много изменений, есп работает на другой частоте, может быть, чайник не признает ее за старое устройство. Хотя возможны и другие причины, конечно.

Toropyga, 2022-07-28:
Format NVS выполнял, ставил версию на частоте 160MHz. Ни как не хочет.
Лог, как обещал:
[r4sGate_2022.07.29_v2022.07.25.log](https://github.com/alutov/ESP32-R4sGate-for-Redmond/files/9213466/r4sGate_2022.07.29_v2022.07.25.log)
Вот выдержка из него:
```
 [0;32mI (6823) R4S: Searched 1 device MiKettle
 [0m
 [0;32mI (6903) R4S: Scan stop successfully [0m
 [0;32mI (6903) R4S: Connect 1 to the remote device [0m
 [0;32mI (7343) R4S: CONNECT_EVT 1, set no Bond and no Encryption [0m
 [0;32mI (7343) R4S: Open 1 success, conn_id 0, if 1, status 0, mtu 23 [0m
 [0;32mI (7343) R4S: Remote BD_ADDR: b8:7c:6f:88:26:b0 [0m
 [0;32mI (7463) R4S: update connection params status = 0, min_int = 6, max_int = 18,conn_int = 18,latency = 0, timeout = 50 [0m
 [0;32mI (9333) R4S: Discover service 1 complete conn_id 0 [0m
 [0;32mI (9333) R4S: SEARCH RES 1: conn_id = 0x0 is primary service 1 [0m
 [0;32mI (9333) R4S: start handle 0x38 end handle 0x49 current handle value 0x38 [0m
 [0;32mI (9343) R4S: Xiaomi Service1 1 found [0m
 [0;32mI (9343) R4S: SEARCH RES 1: conn_id = 0x0 is primary service 1 [0m
 [0;32mI (9353) R4S: start handle 0x23 end handle 0x30 current handle value 0x23 [0m
 [0;32mI (9353) R4S: Xiaomi Service2 1 found [0m
 [0;32mI (9363) R4S: SEARCH RES 1: conn_id = 0x0 is primary service 1 [0m
 [0;32mI (9363) R4S: start handle 0x10 end handle 0x22 current handle value 0x10 [0m
 [0;32mI (9373) R4S: Xiaomi Service3 1 found [0m
 [0;32mI (9383) R4S: SEARCH RES 1: conn_id = 0x0 is primary service 1 [0m
 [0;32mI (9383) R4S: start handle 0x31 end handle 0x37 current handle value 0x31 [0m
 [0;32mI (9393) R4S: Xiaomi Service4 1 found [0m
 [0;32mI (9403) R4S: Update handle = 0x37 [0m
 [0;32mI (9403) R4S: Register_for_notify 1 [0m
 [0;32mI (9423) R4S: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 23, conn_id 0 [0m
 [0;32mI (9463) R4S: Write_auth 1: [0m
 [0;32mI (9463) R4S: 90 ca 85 de  [0m
 [0;32mI (9533) R4S: Scan starting [0m
 [0;32mI (9533) R4S: Scan start success [0m
 [0;32mI (9653) R4S: Write_auth_mi 1 Ok [0m
 [0;32mI (9803) R4S: Read_auth 1: [0m
 [0;32mI (9803) R4S: 6f 39 52 ec 1f 04 23 2d 07 bc b0 b9  [0m
 [0;32mI (9803) R4S: Invalid 1 Xiaomi product Id [0m
 [0;32mI (9813) R4S: ESP_GATTC_DISCONNECT_EVT, reason = 22 [0m
```

PilgrimF, 2022-07-28:
Format NVS выполнял, ставил версию на частоте 160MHz. Ни как не хочет.
Лог, как обещал:
[r4sGate_2022.07.29_v2022.07.25.log](https://github.com/alutov/ESP32-R4sGate-for-Redmond/files/9213466/r4sGate_2022.07.29_v2022.07.25.log)
Вот выдержка из него:
```
 [0;32mI (6823) R4S: Searched 1 device MiKettle
 [0m
 [0;32mI (6903) R4S: Scan stop successfully [0m
 [0;32mI (6903) R4S: Connect 1 to the remote device [0m
 [0;32mI (7343) R4S: CONNECT_EVT 1, set no Bond and no Encryption [0m
 [0;32mI (7343) R4S: Open 1 success, conn_id 0, if 1, status 0, mtu 23 [0m
 [0;32mI (7343) R4S: Remote BD_ADDR: b8:7c:6f:88:26:b0 [0m
 [0;32mI (7463) R4S: update connection params status = 0, min_int = 6, max_int = 18,conn_int = 18,latency = 0, timeout = 50 [0m
 [0;32mI (9333) R4S: Discover service 1 complete conn_id 0 [0m
 [0;32mI (9333) R4S: SEARCH RES 1: conn_id = 0x0 is primary service 1 [0m
 [0;32mI (9333) R4S: start handle 0x38 end handle 0x49 current handle value 0x38 [0m
 [0;32mI (9343) R4S: Xiaomi Service1 1 found [0m
 [0;32mI (9343) R4S: SEARCH RES 1: conn_id = 0x0 is primary service 1 [0m
 [0;32mI (9353) R4S: start handle 0x23 end handle 0x30 current handle value 0x23 [0m
 [0;32mI (9353) R4S: Xiaomi Service2 1 found [0m
 [0;32mI (9363) R4S: SEARCH RES 1: conn_id = 0x0 is primary service 1 [0m
 [0;32mI (9363) R4S: start handle 0x10 end handle 0x22 current handle value 0x10 [0m
 [0;32mI (9373) R4S: Xiaomi Service3 1 found [0m
 [0;32mI (9383) R4S: SEARCH RES 1: conn_id = 0x0 is primary service 1 [0m
 [0;32mI (9383) R4S: start handle 0x31 end handle 0x37 current handle value 0x31 [0m
 [0;32mI (9393) R4S: Xiaomi Service4 1 found [0m
 [0;32mI (9403) R4S: Update handle = 0x37 [0m
 [0;32mI (9403) R4S: Register_for_notify 1 [0m
 [0;32mI (9423) R4S: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 23, conn_id 0 [0m
 [0;32mI (9463) R4S: Write_auth 1: [0m
 [0;32mI (9463) R4S: 90 ca 85 de  [0m
 [0;32mI (9533) R4S: Scan starting [0m
 [0;32mI (9533) R4S: Scan start success [0m
 [0;32mI (9653) R4S: Write_auth_mi 1 Ok [0m
 [0;32mI (9803) R4S: Read_auth 1: [0m
 [0;32mI (9803) R4S: 6f 39 52 ec 1f 04 23 2d 07 bc b0 b9  [0m
 [0;32mI (9803) R4S: Invalid 1 Xiaomi product Id [0m
 [0;32mI (9813) R4S: ESP_GATTC_DISCONNECT_EVT, reason = 22 [0m
```

P.S.: Простите, забыл переключить аккаунт )

alutov, 2022-07-29:
Поправил. Должно работать. У меня работало потому, что  шлюз 0, позиция чайника 1 и mac authentication  выключен.

PilgrimF, 2022-07-29:
Отлично!
Всё работает.
Спасибо!

closed 2022-07-29 by PilgrimF