alutov/ESP32-R4sGate-for-Redmond #103: MQTT Room support

DmytroMelnyk, 2022-08-05:
Недавно узнал про интеграцию MQTT Room в ХоумАссистанс. Вот [контракт ](https://www.home-assistant.io/integrations/mqtt_room/#usage)который нужен для этой интеграции. На этой же странице есть варианты уже существующих интеграций. Наибольшая сложность как по мне состоит в вычеслении расстояния. В одном из репозиториев нашел вот такую вот [формулу](https://github.com/HeimdallMidgard/ESP-32-BLE-Scanner/blob/fdb8b65a4571fdebe2aed154e8198068864ca5d9/src/main.cpp#L241). Возможно ли реализовать такое?

DmytroMelnyk, 2022-08-06:
Я правильно понимаю что паблишинг данных по BLEMonitor происходит вот где-то [тут](https://github.com/alutov/ESP32-R4sGate-for-Redmond/blob/master/main/r4sGate.c#L8379)?

alutov, 2022-08-06:
Пока сильно в это дело не вникал. Насколько я понял, расстояние вычисляется на основе затухания по пути от метки к шлюзу. Для этого им нужна еще мощность передатчика. Но затухание сильно зависит от скажем так рельефа местности между передатчиком и приемником, и в общем случае нужны серьезные вычисления. Это как я понимаю. Пока что остановился просто на измерении меток несколькими шлюзами.

Psytoshgen, 2022-08-17:
В  сотовой связи расстояние так и измеряется, но для точного вычисления используется так назывемый Хопинг - переброс девайса между секторными антеннами вышек, таким образом мы  можем перебрсывя телефон  между вышками увидеть место  с точностью до метра. без хопинга можно увидеть лишь радиус в котором находится устройство. В остльном все верно по задержке сигнала. Полагаю что самый верный вариант использовать zigbee сеть, заставляя  устройства считать расстояния между друг другом и использовать свой ааналог хопинга.

closed 2023-08-14 by alutov