alutov/ESP32-R4sGate-for-Redmond #105: Поддержка устройства: Xiaomi Mi scale v1 and v2

GigaVik, 2022-08-23:
Здравствуйте!
Прошу рассмотреть возможность добавить поддержку весов от Xiaomi а именно MI Scale и Mi Scale 2.
Подробное описание протокола нашлось на GitHub'е проекта openSacle - https://github.com/oliexdev/openScale/wiki/Xiaomi-Bluetooth-Mi-Scale.

alutov, 2022-08-23:
Спрашивали за весы, думаю, добавлю. За протокол спасибо, почитаю.

GigaVik, 2022-08-23:
Возможно еще рассмотрите добавление датчиков влажности\температуры от тех же xiaomi. 
Здесь описание данных популярной кастомной прошивки для датчика Xiaomi Thermometer LYWSD03MMC - https://github.com/atc1441/ATC_MiThermometer

alutov, 2022-08-24:
Есть уже поддержка этих термометров.

GigaVik, 2022-08-24:
> Есть уже поддержка этих термометров.

Блин, точно я в истории версий только сейчас увидел
Чтобы его добавить нужно выбрать устройство Mi-Unknown?

alutov, 2022-08-24:
Все есть в в описании. Прочитайте его, пожалуйста.

Psytoshgen, 2022-08-25:
можно свои  копеек? Yunmai color - аналог Сяомми
Протокол вот
https://gist.github.com/conoro/f0c1d96c450a8f5cce70e2846c3686c4

alutov, 2022-08-29:
Спасибо за линки, буду изучать.

alutov, 2022-09-08:
По протоколу вроде бы выходит, что весы mi scale отправляют вес в рекламном сообщении. Вариант предпочтительней постоянной связи. Вопрос такой, мак адрес весов не меняется? И как выглядит рекламное сообщение в мониторе шлюза.

GigaVik, 2022-09-09:
> Owner
Мак адрес статичный.
Постоянного подключения нет, весы просыпаются в момент их активации и вещают пока идет взвешивание.
[image]

```

mac: c8:47:8c:ef:9c:4c |   | -82 dB | 0.2 s | 0.4 s | 02 01 06 03 02 1d 18 09 ff 57 01 c8 47 8c ef 9c 4c 0d 16 1d 18 02 20 44 b2 07 01 01 0d 0f 0e | s
-- | -- | -- | -- | -- | -- | --
  | c8478cef9c4c |   | -82 dB/0

```

alutov, 2022-09-09:
Спасибо, то что нужно!

alutov, 2022-09-09:
Не совсем совпадает с протоколом. Но по данным выходит, что время и дата 13:15:14 и 10/01/1970.  А вес 8.720 или я ошибаюсь?

GigaVik, 2022-09-09:
> Не совсем совпадает с протоколом. Но по данным выходит, что время и дата 13:15:14 и 01/01/1970. А вес 87.20 или я ошибаюсь? Дату и время выводить?

без синхронизпции с прилржением там можнт быть любая дата.. а вот вес 87.55 был)

alutov, 2022-09-09:
По синхронизации, если приложение установит время и дату, они там сохранятся же, или там батареи нет? 
Еще вопрос, что делать с данными. К примеру, положили что-то на весы, они включились, передали данные и выключились. В умном доме оставлять последний вес и его дату даже при отключенных весах? Или же чистить данные?

GigaVik, 2022-09-09:
Скорее всего он уходит в глубокий сон, и ещё он вроде бы хранит всю историю взвешиваний с самого первого.
По ссылке вроде бы разбор протокола был и работы весов.
В идеале фиксировать последнее значение, 
Сейчас через бле интеграцию в home assistant данные приходят, но после перезагрузки уходят, я эту проблему решал сперва отправляя вес с весов на мктт брокер с фиксацией сообщения а потом уже в home assistant брал оттуда. 
Чтобы не обнулялись данные.
После этого у меня ещё стоит четыре отдельных сенсора на четыре разных человека, весы измеряют вес - сенсоры определяют в их ли диапазоне этот вес и сенсор каждого человека принимает эту цифру

alutov, 2022-09-10:
Есть вот такая ссылочка: https://www.pangodream.es/read-xiaomi-mi-smart-scale-using-an-esp32. Вроде как правильно расписаны данные, приходящие в рекламном пакете. В папку jpg залил версию, можно пробовать.

GigaVik, 2022-09-10:
> Есть вот такая ссылочка: https://www.pangodream.es/read-xiaomi-mi-smart-scale-using-an-esp32. Вроде как правильно расписаны данные, приходящие в рекламном пакете. В папку jpg залил версию, можно пробовать.

[image]
[image]
[image]
[image]

alutov, 2022-09-10:
Нет, соединение я не делал, только в ble-мониторе.

GigaVik, 2022-09-10:
> Нет, соединение я не делал, только в ble-мониторе. Это новая версия? И что показывает в мониторе?

да, новая вот
[image]

alutov, 2022-09-10:
Ну вроде внизу появилось Mi Scale. А появились топики веса даты и времени? Для этого нужно тайм-аут поставить не равный 0.

GigaVik, 2022-09-10:
[image]
показывает правильно, скажи пожалуйста а там в описании протокола не написано весы вещают зафиксированный вес и не зафиксированый разными битами?
потому что те данные которые ты вывел ловят все изменения весов))

alutov, 2022-09-10:
Так, время без цифр часов. Поправил и залил. Выводятся все данные с ненулевым весом. Вот картинка из ссылки: 
[image]
Бита стабилизации на ней нет. В отличие от всех остальных ссылок, где даже порядок данных иной. Я так понял, везде описывается именно соединение. Но там все можно настроить один раз в родном приложении, и больше соединение не особо и нужно, потому и не заморачивался пока. Но там автор заметил, что стабильные показания идут с правильным годом. Чтобы проверить это, надо смотреть данные в динамике. И еще, это версия vi scale 1 кажется. Во второй версии похоже есть отличия.

GigaVik, 2022-09-10:
> Поправил и залил

нету в jpg

alutov, 2022-09-10:
Да, забыл последнюю кнопку нажать.)

GigaVik, 2022-09-10:
> Чтобы проверить это, надо смотреть данные в динамике. И еще, это версия vi scale 1 кажется.

Сегодня позже протестирую в динамике.
MiScale2 v1 
Есть ещё v2 там данных больше, они сами рассчитывают жир и так далее. И внешне другие немного.

GigaVik, 2022-09-10:
[image]

Хорошо работает
Показывает реальную дату и время по UTC, но только если весы были подключены к  офф приложению после установки батареек.
Если вынуть батарейки то соответственно дата 1970
Время и дату показывает при каждом поступившем сообщении, не важно зафиксировано или нет.
Но ble интеграция в homeassistant  каким то образом видит зафиксированный вес (возможно вес фиксируется по жёстко фиксированной выдержке времени с момента получения измерения и интеграция просто отсчитывает ровно столько же как и весы и происходит фиксация)

alutov, 2022-09-10:
Пока оставляем так?

GigaVik, 2022-09-11:
> Пока оставляем так?

вполне!
а можно добавить к весу единицу измерения kg?
Просто homeassistant не стоит графики без единицы измерения.

alutov, 2022-09-11:
Добавил единицу измерения. Обновил релиз.

GigaVik, 2022-09-13:
> Бита стабилизации на ней нет

я нашел бит стабилизации! это 5й бит 0го байта
можно пожалуйста когда будет время сделать 2 датчика веса?
1 как сейчас а второй только стабилизипованный?
[image]

GigaVik, 2022-09-13:
а еще ошибка такая, при удалении топиков mqtt из настроек шлюза,мак весов меняется на такрй же мак + в концеикуча нулей, и устройство после этого неисчитывается.
я как аоспроизведу это, отпишусь.

GigaVik, 2022-09-13:
1
[image]
2
[image]
1 и 2 как выглядит при первом появленти и добавлении

3
[image]
так после удаления mqtt топиков

4
[image]
и появляется новый сигнал от весов, но уже вне условия добавления (без значения интервала времени)

alutov, 2022-09-13:
uuid поправлю, а вот с весами и 5 битом. Там кроме веса есть еще и дата. Фиксировать дату только для стабильного веса?

alutov, 2022-09-14:
Можно пробовать.

alutov, 2022-09-23:
По Yunmai color забыл отписаться. Протокол почитал, но там речь идет о соединении. Сделать можно, но. Шлюз постоянно поддерживает соединение. Удобно по многим причинам, главная - быстрое исполнение команд. В случае весов постоянное соединение не подходит, соединение по требованию - пока нет у меня понимания, как сделать. Для весов отличный вариант - разбор рекламного сообщения. Возможно, там есть все данные, как и в сяоми. Для начала, можно глянуть, что показывают весы в мониторе шлюза.

closed 2022-10-30 by alutov