alutov/ESP32-R4sGate-for-Redmond #106: Отключается nightlight при кипячении (RK-G200S)

MrC0nst, 2022-09-03:
Если включен режим nightlight то при кипячении он выключается и не включается обратно после того как чайник закипел
Вроде в прошлых версиях данный режим не выключался самостоятельно, по крайней мере в настроенной автоматизации по смене цвета все работало. Андрей, есть ли возможность возвращать настройки nightlight после кипячения в исходное состояние
PS: в выпадающем списке меню нет возможности выключить nightlight, а только включить его
[image]

alutov, 2022-09-03:
В принципе так всегда и было. По памяти, там настраивается 2 режима - подсветка при кипячении и подсветка при выключенном чайнике. При кипячении подсветку шлюз не использует и настройки не меняет, управляется она только при выключенном чайнике. Отдельной команды выключения подветки нет, так как она совпадает с выключением любого другого режима.

MrC0nst, 2022-09-04:
>> Возврат подсветки после выключения любого режима в принципе сделать можно.
Да, ее можно держать всегда в режиме ON, А выключать из умного дома передавая цвет 0,0,0

alutov, 2022-09-05:
Ночник может и не всем нужен. Я, к примеру, им не пользуюсь на 216 чайнике, а использую температурную подсветку.

alutov, 2022-09-13:
Закинул в папку jpg версию. Если включать кипячение из шлюза, после его отработки восстанавливается ночник. Если включать кипячение на чайнике, то он восстанавливает ночник сам. Это на 216 модели.

MrC0nst, 2022-09-15:
Только что прошился на 2022.09.14. Модель не 216 поэтому ночник отключался на версии из JPG. В автоматизации я ночник не отключаю а посылаю на него черный цвет (0,0,0).

alutov, 2022-09-15:
Пока что пробный вариант отключения, только на восстановление ночника при включении кипячения на шлюзе. Ночник отключается и при ручном включении кипячения?

MrC0nst, 2022-09-19:
Итак, чайник G214S, на шлюзе выбран как G200S. Мониторю через MQTT Explorer. Включение nightlight через шлюз. Кипячение чайника через кнопку на нем.
[image]
До 1 подсветка включена, **1**- кипячение чайника с кнопки на нем. После кипячения подсветка работала. **2**- Снял чайник с подставки и налил воды. Подсветка сама отключилась, но потом включилась с аппаратным переливом цветов. Секунд через 30 установился цвет из HA. **3**- снял чайник и поставил рядом с платформой. Потом вернул обратно. Опять подсветка включилась с аппаратным переливом цветов а потом отключилась (начало **4**). Включил nightlight через шлюз

alutov, 2022-09-19:
Прошивка пока что возвращает подсветку завершении кипячения, если кипячение было включено на шлюзе. Если она была включена до этого любым способом. И это пока что все. Если кипячение было включено на чайнике, он затем сам восстанавливает подсветку, без участия шлюза. Замечу походу, что при включении кипячения на шлюзе ему приходится явно выключать режим подсветки, а иначе кипячение никак не включить. А потому сильно сомневаюсь, что подсветка потом возвращалась на любой версии шлюза. При снятии с подставки и возврате поведение определяется только чайником. Да, мой чайник тоже включает подсветку при установке на подставку, потом ее выключает. Почему так сделал производитель, пока загадка.  Принудительное включение последнего режима после установки чайника на подставку пока не реализовано, пока подумываю над этим. Но нужно наверное иметь возможность его отключения.

closed 2022-10-30 by alutov