alutov/ESP32-R4sGate-for-Redmond #108: NTP

ggagnidze, 2022-09-10:
Спасибо за прошивку, работает отлично. 

Есть предложение дергать внешний ntp только если не получил по его по dhcp.
Большинство людей запрещает выход в интернет для IoT устройств. А вот до локального (или внешнего одобренного) NTP — обычно дают. 

P. S. Так же можно пробовать NTP на ip шлюзе.

alutov, 2022-09-11:
Согласен, если есть возможность, система должна работать автономно. С ntp не разбирался пока, просто вызываю процедуру, а она работает по умолчанию. Насчет инета, ну не знаю. Шлюз может тягать картинки и погоду из инета, так что я как-то на него рассчитывал). И да, никаких бэкдоров и прочих заморочек в прошивке нет, ну, кроме тех, что есть в в самой среде esp-idf.

alutov, 2022-09-11:
Покопался в документации, есть возможность установить ntp сервер. Если в настройках добавить это поле, думаю, будет достаточно?

ggagnidze, 2022-09-11:
Думаю да. А там в мануалах нет «получать ntp по dhcp»?

alutov, 2022-09-12:
Нашел птичку в sdkconfig, но что-то шлюз не получает время, если ему запретить доступ в инет. У меня кинетик, что там идет по dhcp, непонятно. В гуи нет ничего, в cli пока не разобрался. Вот только не понял, по dhcp идет время или только список серверов.

ggagnidze, 2022-09-12:
У мен список серверов. Типа так же как dns, например. Наверное не все выдают. У меня микротик, там это настраивается.

alutov, 2022-09-13:
Закинул версию с установленной опцией dhcp в папку jpg.

ggagnidze, 2022-09-13:
Так. Пока чего-то не синхронизирует время. Можно какой-нибудь отладчик? Хотя бы чтобы показал какой сервер выхватил из dhcp

alutov, 2022-09-13:
Почитаю еще по отладке, что-то там есть.

alutov, 2022-09-13:
Смотрю лог, получается, что роутер выдает внешние адреса. Если доступа в инет нет, то и время есп не получает. Выходит, что только ntp на роутере. О котором знает dhcp.

ggagnidze, 2022-09-13:
Все так. dhcp настраивается на выдачу конкретного ntp (если прошивка роутера / dhcp сервера позволяет). Он может быть и локальные и внешний, главное чтобы роутер фаерволом его разрешал. 

В Микротике, например, это настраивается так https://настройка-микротик.укр/wp-content/uploads/nastrojka-mikrotik-snpt-client-ukazat-server-vremeni-v-lokalnoj-seti.jpg

А эти логи в консоли по uart видно? Или надо как-то отдельно смотреть?
Кстати, неплохой вариант вытащить консоль прямо в браузер, как сделано у esphome, например.

alutov, 2022-09-13:
В релиз от 2022.09.13 добавил в настройки возможность выбота сервера NTP. Флаг DHCP отключил.

closed 2022-10-02 by alutov