alutov/ESP32-R4sGate-for-Redmond #115: Чайник RK-M216S

Professor338, 2022-11-03:
Добрый день. 
Подключал сегодня чайник "RK-M216S" по инструкции но он подключился как то не полноценно, в отличие от ваших скриншотов у меня поле state пустое, а поле status в MQTT всегда offline....

[image]

и на вкладке подключения что то идёт не так почему то "Device not defined. " и не понимаю что нужно сделать "Store values then press Ok."

[image]

alutov, 2022-11-04:
Нужно зайти в настройки и выбрать в "TYPE" "RK-G200S" или же  "RK-G240S", без разницы. Эти пункты абсолютно одинаковы, просто задают интервал чайников 200-240, для которых выбирается этот протокол.

Professor338, 2022-11-04:
Ура! Спасибо большое!

closed 2023-03-19 by alutov