alutov/ESP32-R4sGate-for-Redmond #117: Не сохраняется установленная температура при включении auto

sbasmanov, 2022-11-15:
В сущности climate, которая появляется в hass после настройки чайника, имеется регулятор температуры (target temp) и режим работы (hvac mode) auto, heat, cool, off. При включении режима auto, установки температуры и последующего выключения, сбрасывается установленная температура (становится 0,0). В чем тут основная проблема: если управлять чайником из homekit, то там установленная температура сохраняется и при последующем включении, отображается последняя настроенная температура, тогда как по факту, в этот момент в HA температура стоит ноль. Это происходит только при переходе из off в auto. При включении heat, температура восстанавливается корректно. Проверил как через homekit, так и через интерфейс HA.
Еще одна странность - в режиме auto или heat если установить температуру 95 или 100 - чайник просто выключается. Это так и должно быть?

alutov, 2022-11-15:
При включении кипячения, а точнее, при выключении режима подогрева температура шлюзом сбрасывается в 0.  Сделано для единообразия и тянется со времен чайников со старым протоколом вроде 173, где для выключения подогрева посылается нулевая температура. Тогда и пришла идея хранить в памяти шлюза последнюю установленную температуру, и даже есть возможность установить ее по умолчани в настройках. Это дает возможность, кроме прочего, восстанавливать ее при установке чайника на подставку. Пишу по памяти, может что нужно будет уточнить. Можно еще подумать, конечно, как все же не сбрасывать температуру в новых моделях, но пока так. Что касается температуры. Шлюз ограничивает температуру подогрева от 35 (раньше было 40) до 95. Сделано из-за того, что некоторые чайники не совсем корректно отрабатывают неверную температуру. Скажем, если на 173 чайнике (у него диапазон 40-80) установить 30, то он ставит 80, что не совсем корректно. А, скажем, 216 позволяет выставить и 35.  Нижнюю границу сдвинул, так пока и осталось.

sbasmanov, 2022-11-16:
Ага, логика понятна. И в принципе, это работает, если использовать только hass. В homekit пробрасывается только термостат. Можно, конечно, пробросить отдельные выключатели, но это будут уже отдельные сущности на разных карточках, никак не связанные с чайником. Я пробовал одну интеграцию в hass для r4s, так там режим boil включался, если установить температуру 100.

>Сделано из-за того, что некоторые чайники не совсем корректно отрабатывают неверную температуру. Скажем, если на 173 чайнике (у него диапазон 40-80) установить 30, то он ставит 80, что не совсем корректно. А вот 216 позволяет выставить и 35. Нижнюю границу сдвинул, так пока и осталось. 

Но с другой стороны, там же выбирается модель чайника при настройке - значит можно по идее определить границы для каждой модели? Для старых поставить 40-80, для новых 35-95. Либо вообще, задавать их в конфигурации. При попытке со стороны hass установить температуру за пределами диапазона, ставить её по соответствующей границе.

>Что касается верхней границы, можно включать кипячение, но опять же, это уже другой режим, проще просто выключать чайник. И еще один момент. Если включать кипячение, когда чайник выключен, включается просто кипячение. Если же кипячение включается, когда чайник в режиме подогрева, то включается кипячение с последующим подогревом.

Вот с кипячением я пока не могу понять, как сделать, что бы было удобно при помощи только термостата. Надо подумать, есть пару мыслей.

alutov, 2022-11-17:
Можно сделать включение кипячения при установке температуры  ровно 100. Но тогда по той же логике нужно при включенном  режиме кипячения выводить температуру 100. Раньше тоже думал об этом, но не делал, так как не хотелось выводить не то, что на самом деле выдает чайник.

sbasmanov, 2022-11-17:
Тут вопрос в логике действий через сущность climate. Если использовать дополнительные кнопки boil/heat в hass, то проблем, в общем-то, никаких. А вот если это передается в homekit, то там уже надо думать. По сути, надо реализовать три режима: нагрев, кипячение+нагрев и просто кипячение. И для этого у нас есть всего два режима - heat и auto. C нагревом всё просто - ставим heat, ставим температуру - он греет. С кипячением сейчас тоже вроде просто - ставим auto и включается кипячение, при этом температура стоит 0. Вот теперь вопрос, как реализовать кипячение+нагрев. Можно в режиме auto выставлять температуру после включения - тогда по идее он сначала закипятит, а потом должен перейти в heat на установленную температуру (так и не проверил до сих пор - так ли это). Но тогда возникает неудобство, что не сохраняется температура (что особенно сильно бесило в родном приложении) и homekit показывает неправильно. При этом, если сделать сохранение температуры, то тогда невозможно будет просто закипятить - чайник будет потом переключаться в режим нагрева. Тогда остается вариант, что auto - с сохранением температуры, это будет кипячение+нагрев, а если надо просто закипятить, то тогда включать boil при выборе 100 (ползунок просто в край) из любого режима (кроме off). Выложу скрины, что бы было понятно, о чем речь.
Выключено, изменять температуру нельзя:
[image]
Нагрев до 80:
[image]
Если после нагрева перевести в auto, то температура 80 сохраняется и включается режим boil + heat:
[image]
[image]
А вот если после этого выключить и обратно включить auto, то температура сбросится на 0 и останется только boil. При этом homekit все равно будет показывать 80 (как на скрине выше):
[image]
Такое впечатление, что homekit не воспринимает сброс температуры на 0 при выключении. Если посмотреть через homespy, то у target temperature min 7, max 100. Не знаю, откуда он берет эти значения.
[image]

sbasmanov, 2022-11-17:
Либо можно оставить всё как есть, просто в документации указать на этот момент - для включения режима кипячение+нагрев, сначала включаем нагрев, выставляем температуру и потом просто переключаем в auto. Тогда он закипит и будет поддерживать указанную температуру. В принципе, тоже нормальный вариант.

alutov, 2022-11-17:
Только добавлю ползунок на 100 - кипячение или кипячение и подогрев. Просто дублируется auto. И если только кикипячение - то целевая температура 100 до завершения кипячения.

sbasmanov, 2022-11-17:
А, еще момент, пока не забыл. У climate есть атрибут hvac_action - показывает текущее состояние нагревателя. Значения heating / cooling / idle. Не знаю, есть ли этот атрибут у mqtt. В принципе, просто красивая плюшка, показывает что сейчас происходит. В homekit отображается стрелкой вверх/вниз или минусом (idle).

sbasmanov, 2022-11-17:
>Только добавлю ползунок на 100 - кипячение или кипячение и подогрев. Просто дублируется auto. И если только кипячение - то целевая температура 100 до завершения кипячения. Вроде логично.

Так если пользователь ставит 100 - тут без вариантов, только кипячение. Не может же быть подогрев на 100 ? Т.е. если 100 ставят в режиме heat - включаем boil, выключаем heat, ставим auto. Если в auto ставят 100 - ничего не делаем, boil и так уже включен.

closed 2023-08-17 by alutov